Akaflieg-Freiburg/enroute #204: Flight direction arrow disappeared after creating and loading a route

hbeil, 2021-09-23:
Android 7 

The app was working fine  until I played with routes. 
As a side note I cannot use the direct to function anymore. 
The app reduced to a mere moving map with airspace and frequency information. 

habe die App dann neu installiert ohne Veränderungen .
Vielleicht ist es ja nur eine verklemmte Setting. Aber Settings erlaubt kein general "reset to Factory settings" oder so

Ähnliche Apps zeigen den Richtungspfeil auf dem gleichen Gerät. 

Gibt es hier eine Lösung, außer den Reset des Mobile Phones?

Beispiel im Flug:

[image]

kebekus, 2021-09-24:
Dear hbeil,

from your description and from the screenshot, it seems that the problem has nothing to do with the flight route that you mentioned. The problem is that the app does not receive any positioning information from the Android system -- this is indicated for instance by the gray dot inside the moving map. Open the main menu and go to Information/Positioning for more information. This is also why "direct to" does not work.

I cannot tell from the distance why positioning does not work. Here are a few random guesses:

- You are inside a building and your sat receiver is not strong enough. -> Spend a few minutes outside to check (older versions of the app also used network information for position estimates, but newer versions use satellite only)
- Positioning is switched off in your device -> Check settings, switch it on again.
- Satellite navigation is broken in your device -> Try with another device to check. Note that other apps might still use the network for position estimated.

I am closing this issue now, because this does not seem like a bug in the app. Feel free to re-open if more information is available.

Best,

Stefan.

it seems that the app does not recieve any positioning information.

hbeil, 2021-09-24:
Ähnliche Apps zeigen den Richtungspfeil auf dem gleichen Gerät.... zu gleicher Zeit.

Das heißt andere Apps können das, was Enroute plötzlich nicht mehr kann. Der angehängte Screenshot zeigt einen Flug in ca 2000 ft bei 95 kn. Die Moving Map moved ja mit, deshalb kann ich Positioning switched off ausschließen.
Ich will hier nicht meckern, sondern helfen ein vielleicht öfter auftretendes Problem zu lösen.

Ich werde mal die settings innerhalb Android checken. Bin mir aber sicher, die LOCATION info is on und Enroute allowed, wie sonst würde Enroute meine Position wissen uns die Karte verschieben?

Hartmut

kebekus, 2021-09-26:
Lieber Hartmut,

Ok, verstanden. Mir war nach der ersten Nachricht nicht klar, dass das Problem im Flug auftritt. Ich hatte vermutet, Du probierst die App vom Sofa aus.

Dies ist jetzt ein sehr ungewöhnliches Problem, das ich so noch nicht gehört habe. Ich kann mir im Moment keinen rechtene Reim machen. Könntest Du die App bei nächst Gelegenheit noch einmal im Flug ausprobieren und folgendes nachschauen:

- Welche Programmversion verwendest Du genau?
- Ist die App mit einem Verkehrsdatenempfänger (z.B. FLARM) verbunden?
- Ist der Punkt, den Du für die "Eigene Position" siehst, blau oder grau?
- Was wird unter "Menü/Information/Positionsbestimmung" und "Menü/Information/Verkehrsdatenempfänger" angezeigt? -- Am besten zwei Screenshots.
- Tritt das Problem auch dann auf, wenn Du ein anderes Gerät verwendest?

Grüße,

Stefan.

hbeil, 2021-09-28:
Hallo.
Verwendet wird Version 2.11.2 - siehe Screenshot
Kein Flarm verbunden
Grauer Punkt - siehe Screenshot
Keine Position wird im Menü angezeigt, aber Das Programm, wie auch alle anderen Programme auf dem Unit zeigen Position und Richtung
Andere Geräte haben das Problem derzeit nicht
[image]
[image]
[image]
.

hbeil, 2021-09-28:
Ist schon komisch - das Programm sagt keine Positioning Information, jedoch lokalisiert mich direkt in Schönhagen Flugplatz. Ich konnte bei dem letzten Test nicht fliegen, sondern bin auf dem Flugplatz hin und her gewandert...

hbeil, 2021-10-07:
Wie löse ich nun das SLL handshake Problem?

kebekus, 2021-10-08:
Wäre es möglich, dass ich Du mir Dein Telefon ausleihst? Wenn ich das Gerät in der Hand habe und per "Android Developer Bridge" debuggen kann, kann ich den Fehler vielleicht finden. Ansonsten sitze ich vor meinem Schreibtisch, stochere im Nebel und bin relativ ratlos.

Ich verstehe, dass das evtl etwas viel verlangt ist ...

hbeil, 2021-10-09:
> Lieber Hartmut,
>
> ganz herzlichen Dank! Meine Adresse:
>
> Stefan Kebekus
> Wintererstr. 77
> 79104 Freiburg
>
> Hoffen, wir dass ich das Problem dann relativ zügig eingrenzen und beheben kann!
>
> Grüße aus dem Schwarzwald,
>
> Stefan.

It’s in the mail. Kommt hoffentlich schon Anfang nächster Woche. DHL Nr. 358315857549
Ich habe das Phone zurückgesetzt. Dann kannst du deinen eigenen Google account nehmen. Nach meinem Reset vorgestern ging die Verbindung zum GPS immerhin – aber der SSL Fehler wird dir auch begegnen.
Viel Glück.

Hartmut

kebekus, 2021-10-14:
Kann SSl-Fehler reproduzieren, habe gleich noch anderes Problem gefunden und gefixt.

hbeil, 2021-10-15:
> Das Problem tritt auf Geräten mit Android <= 7.1 auf, wenn die Hardware-Hersteller veraltete Sicherheitszertifikate (die für die Herstellung von sicheren Internert-Verbindungen notwendig sind) nicht aktualisiert haben. Am 30. September ist das relevante Zertifikat abgelaufen, sodass keine sicheren Verbindungen mehr möglich sind. Die Aktualisierung kann nur durch den Hardware-Hersteller erfolgen.
>
> Als Notmaßnahme habe ich eine Umgehungslösung gebastelt, die mit Programmversion 2.12.1 vermutlich ab heute Nachmittag bei Google verfügbar sein wird. Damit läuft die App erst einmal wieder. Könnt Ihr bitte prüfen, ob das Problem damit gelöst ist und mir Rückmeldung geben? -- Danke!
>
> @hbeil<https://github.com/hbeil> Ich würde das Gerät gern noch einige wochen behalten um eine etwas solidere Lösung zu basteln. Ist das in Ordnung?

“würde das Gerät gern noch einige wochen behalten”

@Stefan – klar geht das in Ordnung.

Hartmut

hbeil, 2021-11-28:
> Nutzer melden, dass das Problem in Version 2.12.1 behoben ist. Ich lasse den "Issue" offen, bis ich das Gerät zurückgesendet haben.

Hallo Stefan.
Das Ende des Jahres naht. Brauchst du noch lange das Gerät?
Ich würde gerne zu Ende des Jahres die Dinge geordnet haben.

Danke

Hartmut

kebekus, 2021-11-28:
Lieber Hartmut,

danke für die Nachricht. Und danke für das Ausleihen des Gerätes. Ich bin mit deinem (und einigen anderen) Geräten fertig und habe mit den Rücksendungen begonne. Dein Gerät liegt schon auf meinem Schreibtisch. Kannst Du mir bitte kurz die Adresse nennen?

Grüße,

Stefan.

hbeil, 2021-11-28:
Alles Klar Stefan, hier meine Postadresse

Hartmut Beil
Billy-Wilder Promenade 47
14167 Berlin

Hartmut

hbeil, 2021-12-01:
Stefan - das Gerät ist gestern gut angekommen. Und ich habe gleich mal die app geladen. So weit keine Probleme.

Grüße Hartmut

kebekus, 2021-12-02:
Lieber Hartmut,

ich bedanke mich für das Leihgerät! Da die Probleme wohl behoben sind, schließe ich den Report jetzt.

Bis zum nächsten mal!

Stefan.

closed 2021-12-02 by kebekus