Akaflieg-Freiburg/enroute #211: Configurable IP adress and port for FLARM data

PatrickKutschat, 2021-11-09:
I would like to attach our existing Flarm via an serial WLAN bridge based on a ESP8266. ESP-Link project provides all functionality but uses a fixed IP and port configuration, 

Is it possible to make the IP and port setting for the FLARM connection configurable?

**Describe alternatives you've considered**
Extending current solution to 192.168.4.1 and port 23 would also work

**Additional context**
https://github.com/jeelabs/esp-link

kebekus, 2021-11-14:
Thank you for your suggestion. I am presently extremely busy and need to make up my mind … please give me a few days.

kebekus, 2021-12-13:
Related request…

-------- Weitergeleitete Nachricht --------
Betreff: 	Enroute FlightNavigation, Issue Report
Datum: 	Tue, 21 Sep 2021 12:00:30 +0200
Von: 	Martin Sxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
An: 	xxxxxxxxxxxxxxxxxx

Hallo Markus,

es wäre klasse, wenn noch eine ...

UDP-Verbindung zu Port 10110

... zu den bereits vorhandenen Wifi-Verbindungen dazu kommen würde.

Echt tolles Programm, weiter so :-)))

Viele Grüße Martin

requa3r0, 2022-03-15:
In addition to above, it could be very cool if the app could listen to the OGN radar app, that also broadcast the data locally on a specific port.
Various navigation software like XCsoar etc. can listen to this data, and it works well, if you have mobil phone data service.

XCsoar connects to the OGN radar service on localhost port 4353

https://play.google.com/store/apps/details?id=com.meisterschueler.ognviewer&hl=en&gl=US

kebekus, 2022-03-16:
Dear requa3r0,

your message arrived, and I would like to have the feature that (both of) you describe. However, unless someone helps me with the implementation, it will be a very long time before anything will happen…

Best,

Stefan.

kebekus, 2024-06-02:
Dear @requa3r @PatrickKutschat , the feature that you requested has been implemented in Version 2.31.01. This version is available at Google Play for beta testers. Please try it out, I'd love to hear your feedback.

Best, Stefan.

kebekus, 2024-06-10:
Implemented in 2.31.02, which appeared in Google Play today.

closed 2024-06-10 by kebekus