Akaflieg-Freiburg/enroute #214: Last installed language is selected for UI

mikezulugolf, 2021-12-30:
On an android where more than one language is installed, the app seems to select the last language installed as the language for it's UI. When I have "German" installed as the first language and then add "French" as the second and then install the app, it uses French as the language for the UI and for other text.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Settings, system, languages 
2. add french as a second language
3. install Enroute
4. Start Enroute and check what language it uses

**Expected behavior**
My expectation would be that it uses the first installed language

**Smartphone (please complete the following information):**
 - Device: Nexus 6P
 - OS:Android 8.1

kebekus, 2021-12-31:
Dear mikezulugolf,

I understand your request and I am able to reproduce that issue that you describe. I am, however, unsure how I can possibly fix the bug -- it seems that the issue lies with some of the libraries that I am using.

I will investigate, but I that might take some time…

Best,

Stefan.

kebekus, 2023-01-29:
@mikezulugolf Thank you for the report! It's been a long, long time … but I think the bug is fixed with the last update of the app, Version 2.23.0.

Thanks again … I can only ask for your understanding that I work on this app alone, which implies that things sometimes take a very long time indeed.

Best,

Stefan.

closed 2023-01-29 by kebekus

mikezulugolf, 2023-01-29:
Never mind - this bug was never a showstopper and I can confirm that it was most likely not caused by your code, as I couldn't find any pointers to the potential language selection step anywhere in your code when I reviewed it a year ago..
Thanks a lot for this fantastic app.
Always blue skies,
Richard.