Akaflieg-Freiburg/enroute #216: Smarter way of adding waypoints

kebekus, 2022-01-02:
The app could ask where in the route waypoints should be inserted (at the beginning? at the end? in the middle?)

kebekus, 2022-03-19:
TTC70 wrote: I noticed that when you add a waypoint, the app always adds it after the last waypoint in the flight plan (destination). Is there a way to add a waypoint after the closest one? Or let the user choose where he wants to load the new waypoint?

There is presently no such way. I have not found the time to implement anything here.

@TTC70 Let's use this issue to discuss issues related to 'waypoint adding.'

TTC70, 2022-03-19:
@kebekus ... "I have not found the time to implement anything here.".... of course. I believe that everything should be done in the right time.
Regarding this issue, when I add a new waypoint to my GPS Garmin in the car, the system asks after which waypoint I want to put the new waypoint.
The same concept is applied to the programming of the on board computer of A320: when you have to add a waypoint between the departure airport and the arrival one, you select the departure airport, write the name of the waypoint (VOR, another airport, NDB, TACAN, etc...) and the system writes that waypoint after the one you selected.
This might be an idea....?!

closed 2022-06-22 by kebekus

TTC70, 2022-06-23:
> I just implemented an improvement for the upcoming version 2.18.0. Please have a look once this is out, then we can discuss it further.
> 
> Best,
> 
> Stefan.

Roger. 
I will try the new features and will let you know.