Akaflieg-Freiburg/enroute #217: Data entry  bug in the Aircraft page

bob-g4bby, 2022-01-03:
When 'Units' is set to 'Statute Miles', the True Airspeed values entered are altered by the program after entry is completed

Steps to reproduce the behavior:
1. Go to the 'Aircraft' page
2. Set 'Units' 'Horizontal' to statute miles
3. Select the 'True Airspeed' 'Cruise'
4. Enter a cruising speed
5. Press 'enter' and observe the displayed speed has been slightly more than doubled
6. The same will happen with 'Descent' or 'Minimum'

Expected behavior
I would expect the entered numbers in miles per hour to remain unchanged after 'Enter' is pressed. It looks like they are being multiplied by 2.2, the conversion factor for mph to kph.

Smartphone
 - Device: Motorola G5 Android phone
 - OS: Android 11

Additional context
If the 'Horizontal' Units are kilometres per hour or knots, then the entered data in 'Cruise', 'Descent' and 'Minimum' does not get altered. The fault may be restricted to the Statute Miles setting.

**NB Spelling mistake** - The label 'Minumum' on the 'Aircraft' page should be 'Minimum'

Thank you for lowering the Flight Path Vector speed threshold, that will work with Paramotors now.

kebekus, 2022-01-03:
Dear  bob-g4bby,

thank you for your report. I can reproduce the issue and will work on a solution.

Best,

Stefan.

kebekus, 2022-01-04:
Dear bob-g4bby,

again, thank you for your report. I was able to find the issue fairly quickly and (hopefully) fix it. I just uploaded a fix to Google Play, which should be available in a few hours from now. Could you please check out the latest version (=2.14.4) and confirm that the problem has been solved?

Version 2.14.4 is presently only available for beta-testers. If you are not a beta-tester already, please use the link below to join the beta program (you can leave anytime, of course)

- https://play.google.com/store/apps/details?id=de.akaflieg_freiburg.enroute

Looking forward to your feedback!

Best,

Stefan.

bob-g4bby, 2022-01-04:
I confirm the problem (Issue #217) is solved, Stefan.

 
 

   
   - https://play.google.com/store/apps/details?id=de.akaflieg_freiburg.enroute

kebekus, 2022-01-05:
Thanks again. I am closing the issue now.

Best,

Stefan.

closed 2022-01-05 by kebekus