Akaflieg-Freiburg/enroute #221: Custom maps - Danish ICAO map

requa3r0, 2022-03-05:
Hi

When soaring in Denmark, the gliders use special defined soaring zones.

They are defined on the icao chart here.
https://aim.naviair.dk/da/kort/

For Sjælland 
https://aim.naviair.dk/media/1153/edition-41-icao-anc-1-250-000-copenhagen-area-pdf-300dpi.pdf 

The PDF is Geo-located and your position on the map can be seen for example using avenza maps app, as a viewer. 

Unfortunately, The soaring zones does not load in the maps generated by Enroute flight navigation.

Could it be possible to load data from a local source in the app? 

Either the zones as a layer or perhaps the official geo ref. ICAO map. This map is what most pilots know, and it would be great as a background map 😎

A solution to this problem would be great, both for VFR aircrafts, TMGs and Soaring pilots who just want a moving map to indicate thir position in the zones etz and heading with the time to enter next zone etc. 

Your app would be a very big help for aviation navigation in Danmark if this was solved somehow. 

Thanks for the good work

kebekus, 2022-03-07:
Dear requa3r0,

gliding zones for Denmark are available and should be shown. Can you please check the following:

- Are you using the latest version of the app, 2.15.1?
- Are you hiding gliding zones? Go to the 'Settings' page and check the appropriate switch.

Looking forward to hearing from you.

Best,

Stefan.

requa3r0, 2022-03-07:
Ok.
True.. I was hiding the gliding zones because they don't really make sense.

Its just yellow zones with Yello lines.

They all have names and altitudes

Like N4E..2500-4000feet

This is important information when changing zones with ATC etc. 

Its pretty hard to figure out from the map layout in the app.

The official ICAO does present a lot of information.

Could the zones be present in a way more similar to the icao map and with the Meta data somehow? 
[image]
[image]

kebekus, 2022-03-08:
Dear requa3r0,

> Its just yellow zones with Yello lines. They all have names and altitudes, and are shown with red dotted lines and red text. Like N4E. Above 2500 feet normally opened to 4000 feet. This is important information when changing zones with ATC etc.

I understand what you mean. The names of the glider zones are present in the app, but they are not shown on your device because the map layout algorithm found that your display is too small to display all the data. Altitudes are not normally shown in the map, because my experiments showed that again most maps are too small. You can obtain complete airspace information by double-tapping (or tap-and-hold) on any position. 

To be honest, the app is not optimized for gliding. Personally, I do fly gliders as well as motor planes, and the app was designed to help me with airspace and navigation on long-distance motor flights. In the glider, I typically rely on LX gliding computers, and use the app only now and then.

> The official ICAO does present a lot of information, a bit more straightforward. Could the zones be present in a way more similar to the ICAO map and with the meta data somehow?

Tricky. 

- The design of the moving map in the app follows the ICAO standard map, where gliding areas are yellow. The maps that you show are not typical of ICAO maps. 
- To find a map layout algorithm that works well across a large selection of screen sizes and font sizes (and supports map rotation…) is well-known to be a near-impossible task. The present algorithm is the best compromise that I could come up with.

The app is free software and 100% open source. Perhaps you would like to join the development and help to improve the map? Any input is greatly appreciated!

> Or... What about just giving the user the option to change the background map to the official ICAO map.

This has been on my radar for a while, but I have not yet found the time to implement this. Again, would you be willing to help with the implementation? 

Best,

Stefan.

requa3r0, 2022-03-08:
Thank you for your understanding reply to my request. 
Unfortunately I don't have any knowledge about app programming.
But its nice to know that my feedback has been evaluated.

Keep up the good work.

Compared to other free android based pilot navigation apps, this is very promising indeed ;O)

If I were to give some general feedback, I would say that the Navigation info / feedback needs to be improved, to better suite a cockpit environment, and give the necessary info.

Larger and more clear numbers, and a proper additional Heading information at the top.
Perhaps a rolling dial at the top like the moving band on a Garmin G5 ;O)

I think the information bar at the button should include to bars with 3 numbers each.
Any smartphone today is so high, it makes good sense to use some the aspect ration for NAV info.

Ground speed
Current course
ETE to destination
ETA to destination (perhaps end of flight plan)
Dist to next wpt
ALt MLS

Generally 

If a pilot uses this app, its properly because he dos not have a Garmin G5 or any other fancy instruments.
Perhaps he is flying a vintage aircraft, a TMG or just some old beat up C172 without much more than a standard mechanical gyro manually corrected with a compass, but drifting all over the place.

In all cases its a shaky bumpy ride with bright sunlight, where small numbers on a smartphone are difficult to make out.
The best is really big fat white letters on dark background.

If your given a specific heading by ATC, the roiling dail to find heading 220 is just the way to go ;O)
All pilots understand this, and it would make it easy ;O)

Sorry for going on an on...but that is my 2 bits ;O)

kebekus, 2022-05-22:
@requa3r0  Thank you for your report. The feature that you requested (=custom base maps) is implemented. I have uploaded v2.16.90 to the Google Play Beta Channel. If you are a beta tester, please have a look. I appreciate your feedback!

closed 2022-05-22 by kebekus