Akaflieg-Freiburg/enroute #222: Set wtp as home/destination. Rubber band track

requa3r0, 2022-03-05:
On some flights an actual flight plan doesn't make sense, and setting a direct to.. Does not make sense either.

An example could be if you are flying out of an airport to the vest to do aerobatic.

During the flight it would be nice to be able to set the home airport as home or destination, and the line to the wpt would follow you like a rubber band, and nav info regarding distance and time would update live.

Another example would be just a standard joy ride out to the coastline etc, and your track direct back home would follow you all the time, and navigation info regarding the flight home would update likewise. 
No matter how you choose your route back... Always have the line direct home. 

This could be implemented as a Set to home.... Feature... To complement the direct to.. Which locks the line and give you a specific track to follow.

Just an idea.

It would be great for all the pilots who doesn't fly specific routes but mostly just fly and enjoy the ride ✈️

kebekus, 2022-03-07:
Dear requa3r0,

Thank you for your suggestion. I think I like the idea, and will probably implement the feature for in of the next releases (probably 2.17.0, but I do not promise).

Best,

Stefan.

requa3r0, 2022-03-08:
Thanks for looking into this.

The idear is obviously not new.

All glider navigation software have such a feature, to give the pilot arival information regarding time and altitude predictions etc.

But the feature could very well be implemented into VFR nav app as well, because flying is so much more than straight lines 😎

kebekus, 2022-03-13:
Implemented in upcoming v2.16.0. Use "direct to" to set a destination. As long as you are >3nm away from the destination, a rubber band will be shown. Thank you for the suggestion!

closed 2022-03-13 by kebekus

kebekus, 2022-03-15:
Dear all,

The next release of **Enroute Flight Navigation** is coming up, and much has changed internally. I expect a longer-than-usual beta test phase. I would greatly appreciate your feedback if you could try the new features. I am particularly interested to learn if the app reliably recognizes the next waypoint when you follow a route (critical cases might include: routes with sharp angles, return routes, switching the app on and off during the flight, switching the app off before a waypoint, )

Best wishes,

Stefan.

-----

## New features 

### ETE, ETA, and rubber band 

If you fly along a route, the app will show the remaining distance, ETE, and ETA for the next waypoint and the final waypoint. The moving map shows a red "rubber bad" between the current position and the next waypoint. The information is shown under the following conditions:

- The user has set a route.
- The user has set aircraft cruise speed and wind information.
- The aircraft is flying.
- The plane is closer than three nautical miles to the route.
- The distance to the final waypoint is more than three nautical miles.

### Behavior on direct-to

When double-tap into the moving map, the waypoint description dialog will open, and you have the choice to set a quick "direct to" route. The app will now show the rubber band to the waypoint when you do that. The following conditions must be met:

- The user has set aircraft cruise speed and wind information.
- The aircraft is flying.
- The distance to the final waypoint is more than three nautical miles.

### Map update notifications

The app now uses Android notifications to alert you when map updates are available. The alert will not show up while in flight.