Akaflieg-Freiburg/enroute #223: Cannot save aircraft data

kebekus, 2022-03-17:
[Ivan081](https://github.com/Ivan081) via email 18 hours ago

> Of I set km/h as Speed and save the airplane app do not remember km/h

[TTC70](https://github.com/TTC70) via email 18 hours ago

> I confirm..... it seems the App does not save the new speed unit.
> I tried to change the speed unit, save the airplane, close the app, reopen
> it and the speed unit was still in mph.
> I hope this helps

The issue reports refer to version 2.15.91.

kebekus, 2022-03-17:
I can reproduce the problem here. I will work on a solution.

kebekus, 2022-03-17:
Dear all,

I just uploaded a bugfix version of the app (= 2.15.92) to Google Play, which should by now be available to all beta testers (you might need to manually "check for updates" in the Google Play app).

Perhaps you could have a look and check if the issue is gone?

Beste wishes · Thank you for your help,

Stefan.

TTC70, 2022-03-17:
Hello Stefan.
What I found out so far.
Baseline: Connection with MSFS2020 - plane in flight with flight plan and
wind set up in the app.
- Upper menu with ETA and ETE randomly disappears. Tried to reboot the app,
clean phone RAM and relaunch the app. No success.
  The issue seems to be related to the "connection with the traffic data"
(app set up on traffic data receiver).
- Rubber band appears under described conditions. It disappears though when
the upper ETA/ETE menu disappears.
Tomorrow I will try the sharp turns.
I hope this helps.
P.S.
I noticed that when you add a waypoint, the app always adds it after the
last waypoint in the flight plan (destination).
In a flight plan with, e.g. 10 waypoints, if you want to add a waypoint in
second position, you have to manually move it up 8 positions in the flight
plan.
Is there a way to add a waypoint after the closest one? Or let the user
choose where he wants to load the new waypoint?
Ciao
Antonio

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Mail
priva di virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

TTC70, 2022-03-18:
Stefan, 
I made some attempts with MSFS.
Flight plan was 5 waypoints not so far away (about 5 miles from each other), but the bent were quite tight.
The rubber band comes and goes, apparently randomly from a waypoint to another. The upper menu (the one with ETA and ETE) comes and goes with the rubber band.
The baselines are the same of my previous report.
The only difference is the defined flight plan as stated.
I hope this helps.

kebekus, 2022-03-19:
@TTC70 Thank you for your input! To manage communications, let me try to keep separate things separate. I will open new issues/assign you to existing ones to have one problem per communication thread.

kebekus, 2022-03-19:
@TTC70 @Ivan081 You reported that the horizontal distance unit could not be saved in 2.15.91. Has this issue been solved in 2.15.92, or do you still experience the problem? Thank you for your reports!

TTC70, 2022-03-19:
App Version 2.15.92.
This app version keeps the new saved airplane settings.

kebekus, 2022-03-19:
Thanks! I understand that the problem has been fixed in 2.15.92 and I am closing this issue now. The other issues remain open.

closed 2022-03-19 by kebekus