Akaflieg-Freiburg/enroute #224: Remaining Flight InfoBox randomly disappears

kebekus, 2022-03-19:
TTC70 wrote: Upper menu with ETA and ETE randomly disappears. Tried to reboot the app, clean phone RAM and relaunch the app. No success. The issue seems to be related to the "connection with the traffic data" (app set up on traffic data receiver).

- Rubber band appears under described conditions. It disappears though when the upper ETA/ETE menu disappears.

@TTC70 Let's use this communication thread for issues with the disappearing InfoBox

kebekus, 2022-03-19:
TTC70 wrote: I made some attempts with MSFS. Flight plan was 5 waypoints not so far away (about 5 miles from each other), but the bent were quite tight. The rubber band comes and goes, apparently randomly from a waypoint to another. The upper menu (the one with ETA and ETE) comes and goes with the rubber band. The baselines are the same of my previous report.The only difference is the defined flight plan as stated.

kebekus, 2022-03-19:
@TTC70 I wonder if the problem is related to instability of the connection between the app and your flight simulator. Can you check on the settings page that the "Primary positioning data source" is set to "Traffic Data Receiver" instead of built-in GPS?

TTC70, 2022-03-19:
The source of data is "traffic data receiver".
Thinking anyway about the connection, I will try again today focusing on
the stability of wifi network and will report.

TTC70, 2022-03-19:
Report.
Connection is stable, both internet, wifi and LAN (my laptop is connected
to the internet, and  to the router, via a LAN cable, the smartphone is
connected to the router via Wifi).
I have been trying for 40 minutes (Enroute is the only app running, memory
has been cleaned).
The route is "Z" shaped: Waypoint CA SAVIO, Airport Venezia Lido LIPV,
Venezia Tessera LIPZ, the last waypoint is a user waypoint over Venice city
center.
The flight plan has been selected and activated while in flight.
As previously discussed, the rubber band and the upper menu (apparently)
disappear randomly.
As Michael has said in his answer, maybe the issue is distance related, the
waypoints I choose have the following characteristics:
CA SAVIO -> LIPV - 2.9 miles
LIPV -> LIPZ - 4.8 miles
LIPZ -> waypoint - 3.3 miles.
I chose those waypoints because of their positions (Z shape) and I think
they are ok for a VFR flight.
I will choose another faster airplane with different waypoints and will
report back.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Mail
priva di virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

TTC70, 2022-03-19:
Report.
Connection conditions as per my previous post.
I am flying this very moment with Beach Baron... a bit faster.... i could set up a flight plan with VORs and airports as waypoints.
The behavior of the upper ETA/ETE menu is definitely linked to the rubber band.
As long as i stay on course the ETA/ETE menu (and rubber band) is visible and working properly.
When I try to "fly around" the waypoint out of the 3 miles distance circle, the ETA/ETE menu disappears (as well as the rubber band).
When I go closer to that specific waypoint, the ETA/ETE menu reappears (as well as the rubber band) showing the next waypoint.
The "intensity" of the turn does not change anything: turning 179 degrees or 3 is exactly the same.
I hope it helps.

kebekus, 2022-03-21:
Dear TTC70,

Thank you for your detailed report. The behavior you describe is the intended behavior: The infobox and rubber band are shown if you are flying and closer than three nautical miles to the route. I will therefore close this issue now. Thanks again!

Best,

Stefan.

closed 2022-03-21 by kebekus