Akaflieg-Freiburg/enroute #225: Color of flight route on moving map

kebekus, 2022-04-24:
TTV70 wrote: 

> Stefan, using the app more extensively, I have come to realize that it is more intuitive and of immediate understanding if the flight plan path is magenta.
> Understanding that each programmer has his/her own way to produce software, maybe you can give the user the option to keep your original color of the flight plan path, or choose another color. Or at least between your original color and magenta.
> Magenta is widely used by most of navigators (both for cars and airplanes) and maybe the user might be attracted to the app a little more if there is also this feature.

kebekus, 2022-05-01:
Implemented. I still need to do some beta testing; the change will appear within a few weeks from now. Thank you for your suggestion(s)!

Best,

Stefan.

closed 2022-05-01 by kebekus

TTC70, 2022-05-01:
Well, I'm a beta tester, then i guess I'll see the magenta flight plan as
soon as the update will be available for beta testers.
Cheers.
Antonio

kebekus, 2022-05-01:
@TTC70: Sure. Perhaps even tomorrow