Akaflieg-Freiburg/enroute #226: App does not update correctly

requa3r0, 2022-04-24:
[image]
Hi

I have tried many times to update the app. 
Even the really hard way
Force stop, delete cache, delete storage, uninstall app.
The in the bootloader clear main cache.

When installing the newest version, it's still not working.

It does not open as a fresh install with map downloads etc.

Some data is til present

I have no top bar with the new info
The line for navigation is gone and the update not properly applied.

What do I need to delete to properly wipe the old app.

Look at the screen shot of the newest version.
I have a direct to set!

Kind regards Requaero

kebekus, 2022-04-24:
Dear Requaero,

I am not sure I fully understand, so let me recapitulate

- You are missing the top bar with "remaining route info". Is that correct?
- Apart from the item above, is there any other problem that you have with the app?

In order to help me investigate the issue, can you please look at the main menu and tell me which version of the app you have? Also, if you open the "route" page from the main menu, is there a route that you have set? If yes, can you send me a screenshot of the route window?

Best,

Stefan.

requa3r0, 2022-04-24:
Running latest version 2.16.0 
When installing after wipe of previous app and local delete of empty folder SD card . It did not ask for map download etc. 

Point 1.
Yes no top bar with any info.
Also Not when flying.

If setting a route.. With 2 wpt i get a line.. But not when selecting just one wpt and fly direct to. 

No line and no info on top bar. 
[image]
[image]

requa3r0, 2022-04-24:
[image]

requa3r0, 2022-04-24:
I only see a line with 2 wpt.. Still no top bar. 
NB. I kind of agree with previous issue regarding the line would stand out better in magenta. 

[image]

kebekus, 2022-04-24:
Actually, in your picture there **is** a top bar. It's the text "More than 3 nm off route". This text is shown if are more than 3 nm away from the route. At that distance, you are probably not following the route and it is often hard (or impossible) to guess what your intentions are and how long your flight will take. Move back closer to the route and the text in the top bar will change accordingly.

requa3r0, 2022-04-24:
You are right.
Perhaps my full wipe attempt did solve the issue. 
I have not been able to make the app work for quite some time.

Setting a closer user wpt and then an airport as destination, does give me a route line and a direct line.

Still don't get anylines when just going direct to a point though. 
Perhaps I need to be moving?
Ill test in the car tomorrow.

kebekus, 2022-04-25:
Dear [requa3r0](https://github.com/requa3r0),

ok. It seems that there are three separate issues:

- Uninstalling the app does not wipe all the data. I have opened a new bug report for this issue.
- The "remaining route info" is not what you expect; this seems a non-issue now.
- The route is not shown when going "direct to". This is a bug. I have been able to reproduce this bug on my machines. I have opened a new bug report for this issue and will try to fix this bug as soon as possible.

Thank you for your report! As soon as I have opened the new bug reports, I will close this one and then resolve the issues one-by-one.

Best,

Stefan.

kebekus, 2022-04-25:
Dear [requa3r0](https://github.com/requa3r0),

thanks again. I have opened separate new bug reports for each of the issues that you describe. Please subscribe to them, so I can report progress and contact you.

I am closing this report now.

Best,

Stefan.

closed 2022-04-25 by kebekus

kebekus, 2022-04-26:
@[requa3r0](https://github.com/requa3r0): can you please have a look at the following issues that I opened in response to your request?

- [Issue #228](https://github.com/Akaflieg-Freiburg/enroute/issues/228)
- [Issue #227](https://github.com/Akaflieg-Freiburg/enroute/issues/227)

You need to subscribe to those issues, so we can communicate. Subscription is done via the bar on the right side of the screen, under 'notification'. In each of these issues, can you add a quick comment, so I can see that you are there? -- Thanks!