Akaflieg-Freiburg/enroute #228: Uninstalling the app does not remove all data

kebekus, 2022-04-25:
Users report that uninstalling the app does not remove all the data. When re-installing, previously installed maps are still there.

kebekus, 2022-04-25:
[requa3r0](https://github.com/requa3r0) wrote:

> I have tried many times to update the app. Even the really hard way Force stop, delete cache, delete storage, uninstall app.
> The in the bootloader clear main cache. When installing the newest version, it's still not working. 
> It does not open as a fresh install with map downloads etc. Some data is til present!

kebekus, 2022-04-26:
@[requa3r0](https://github.com/requa3r0): I think I understand the issue now. I believe it's not a bug and that there is no problem. Instead, you experience a feature of the Android operating system called "Auto Backup." Auto Backup is a part of Android that automatically saves app data to a private section of Google Drive, stores it there even when the app is uninstalled, and restores it as soon as you re-install the app. Have a look [here](https://developer.android.com/guide/topics/data/autobackup) for a technical description of Auto Backup.

I understand that users might not be aware that they opted in to "Auto Backup" when they installed Android and might find Auto Backup confusing. I still have to make up my mind if I wish to disable Auto Backup for my app in one of the upcoming versions.

requa3r0, 2022-04-26:
Ok. Makes sense.
Was misled by this feature for sure 😊

kebekus, 2022-05-01:
@[requa3r0](https://github.com/requa3r0): I understand that this feature can be confusing … especially since most people in larger countries will hit the size threshold because their base maps are larger than 27M. I have therefore added code to opt-out of auto-backup. If all goes well, the behaviour of the app will change with the next release.

Thanks again for your help!

closed 2022-05-01 by kebekus