Akaflieg-Freiburg/enroute #230: Allow adding custom airspaces file / location

migajek, 2022-04-25:
I don't know the process for other countries, but currently in Poland we have a number of airspaces that are not published in AIP yet. 
These are defined by the NOTAMs, they are not permanent (at least in theory - there is a notam expiration date provided) so I believe they won't be added to either OpenAIP or OpenFlightMaps.

What I was thinking of is to provide a web service that provides a daily set of NOTAM-based airspaces for Poland. I would implement that myself.
Should anyone else be interested in adding coverage of their country, they could provide a web service with a compatible data format. 
The data returned should include the date of update, the list of airspaces and the list of notams that failed to parse (there are some irregularities in the notams - see below, so I expect sometimes the notam will fail to parse until manual adjustment in parsing code is introduced)

I'm not however sure whether that is possible to display dynamic data on the map, or is everything pre-rendered? 

---

D 202204260900 - 202204261300:
TEMPORARY SEGREGATED AREA EPTS 427 - MILITARY TRAINING.
AIRSPACE CLASS - UNCLASSIFIED.
REAL TIME OF ACTIVATION IS PROMULGATED VIA AUP.

EPTS 427
LATERAL LIMITS:
CIRCLE RADIUS 3,5 NM CENTRED ON POINT: 520300N 0150000E
VERTICAL LIMITS:
GND - 3000FT AMSL
CONTACT: +48 261 685 101, +48 607 318 380

D 202205010000 - 202207312359:
TEMPORARY RESERVED AREA EPTR 446 - MILITARY ACTIVITY.
AIRSPACE UNCLASSIFIED.
LATERAL LIMITS (WGS-84):
1.491633N 0224430E
THEN ALONG STATE BORDER TO THE POINT: _<-- my personal favorite_
2.490543N 0223445E
3.491452N 0223146E
4.491718N 0223232E
1.491633N 0224430E

VERTICAL LIMITS:
7000FT AMSL - FL095
CONTACT - MIL, PHONE NUMBER AVBL IN AMC POLAND

kebekus, 2022-04-25:
Dear [migajek](https://github.com/migajek),

thank you for your report! I understand your request and I am very supportive. To implement everything, I would however need (your) help, because I am writing the app in my spare time, and I am coming to the limits of what I can do.

The feature "Allow adding custom airspaces file" has been requested more than once, usually by pilots/clubs who wish to add locally regulated airspaces to the app (such as noise-sensitive zones, or zones that are available by special agreement). They usually have CUP files that they wish to import into the app -- CUP is not a great formal, but it seems to be the de-facto-standard for aviation data these days. The technology that I use to display maps (=MapBox GL/MapLibre) cannot handle CUP files, but it can handle GeoJSON files. In fact, all the aviation data used by Enroute Flight Navigation comes in the form of GeoJSON, in the format described [here](https://github.com/Akaflieg-Freiburg/enrouteServer/wiki/GeoJSON-files-used-in-enroute-flight-navigation). 

So, what is needed is a CUP-to-GeoJSON converter, implemented in C++, ideally using the Qt library framework. Would you be willing to write that? Once we have this, I could then add the CUP-import-functionality to the app, so that the app would be able to read CUP files coming from your NOTAM-parser, from your local aviation association, or from the owner of the gliding ranch in that you might visit in Namibia.

Dear [migajek](https://github.com/migajek), these are just some random thoughts. I am open to discussion!

Best,

Stefan.

migajek, 2022-04-26:
Hello Stefan,
thank you for the response. I surely understand your limitations, still impressed with the amount of work you're putting alone here. 

Regarding the CUP files, I had a brief look on the format specification as provided here ( http://download.naviter.com/docs/CUP-file-format-description.pdf )  and it seems the format only supports waypoints? I also went through the resources I used to rely on back in the day I did gliding (LK8000, XCSoar) and all the airspaces are provided in the OpenAir format. 

Regarding the implementation - unfortunately I'm not a C++ programmer, I know the basics but didn't use that in ages. So, since I have no fluency in C++, implementing the relatively easy task of parsing text-based format would probably take ages and we'd end up with a code of rather poor quality. Not to mention the lack of Qt knowledge at all.
What I could easily do would be to implement such converter in the technologies I use daily (.net core / C#) - this could be a web service where CUP content is being sent and GeoJSON response is returned. 
Please let me know what's your opinion here :) I surely agree it'd be best to do it offline, but I'm afraid I'm simply unable to provide an implementation that could be incorporated directly into Enroute

kebekus, 2023-06-22:
Thanks to the work of @bloech1, we are making progress:

[image]

kebekus, 2023-07-03:
@migajek Thank you for your suggestion! The feature is implemented and will be included in version 2.28, which will (hopefully) appear by the end of this week.

I am closing this discussion now. Thanks again, and please feel free to suggest more features (or report errors) in the future.

Best,

Stefan.

closed 2023-07-03 by kebekus