Akaflieg-Freiburg/enroute #231: Android reboots sporadically with Stratux connection

christianhbeck, 2022-04-30:
Hello,

i have been using enroute for several flights on a Samsung S2 tab. This was working very well.
Recently I build a Stratux with a softrf. After starting to use the Stratux, the trouble started. On two flights I had one reboot per flight. I did not recognize a specific trigger so far. The reboot did not occur after starting the app (as reported in a closed issue).
A smaller issue I am observing is that when I activate the screen after running enroute before deactivating the screen, the app does not react for ca 30 sec. Feels like something has to run into time out. I am not sure if the two topics are related.

kebekus, 2022-05-01:
Dear [christianhbeck](https://github.com/christianhbeck),

thank you for the report -- I would not be able to find and fix all the bugs without help from our users.

The issue with the reboots is worrying … but I am unsure what to do here. In principle, an app should never be able to crash an Android system. Now, if you do experience crashes, it seems that there is a bug in the operating system, perhaps in one of the hardware drivers. I do not know how to help.

On the other hand, the problem that the app freezes for 30 seconds does seem like a problem that has to do with the app and that I should be looking into. Can you describe a sequence of steps that reproduce the issue? Also, do you perhaps have access to another device, and can you check if the issue appears on the other device as well or if it is device-specific?

Best wishes · Thank you for your help,

Stefan.

christianhbeck, 2022-05-07:
Hi Stefan,
I tried a more up to date android phone. I could not reproduce either of the issues when connecting the Stratux. I apologize for not trying this first.
Thank you for providing this nice software.
Regards,
Christian

closed 2022-05-07 by christianhbeck

kebekus, 2022-05-08:
@christianhbeck No matter what: Thanks for your report. Without input from the users, I would never be able to maintain any  quality standards…