Akaflieg-Freiburg/enroute #232: ETE/ETA not calculating correctly

fly-shadok, 2022-05-01:
In my last flight, ETE/ETA to next WP did not calculate correctly and was consistently about 30min off - see attached screenshot. At 272kph and with 182km remaining, I am shown 76min ETE instead of 40min. I tried clearing and reentering the route, but to no effect.

This may or may not be related to another issue: Sometimes, the route mode showing ETE/ETA does not activate, especially when choosing a new Direct-To in flight. I had to 'cheat' by entering a route consisting of a waypoint behind me and the actual destination to get a track and an ETE/ETA calculation, so maybe it calculated the distance from my position to the dummy WP and then to the destination to establish ETA/ETE?

[image]

Thanks for the app btw, keep up the great work!

kebekus, 2022-05-01:
Dear [fly-shadok](https://github.com/fly-shadok),

Thank you for the report -- I would not be able to spot and fix bugs without help from our users. You are mentioning two issues:

1. The ETE is not computed according to your expectations. 
2. When using "direct to," the moving map does not show the route, and the "remaining route info" does not show the ETE.

Other users have reported Item (2) already. I have (hopefully) fixed the issue. The next version program version will probably appear next week. Please let me know if the problem persists for you after the update.

Item (1) is new. One possible reason: **Enroute Flight Navigation** does not use the current ground speed for ETE computation but the speed that comes out of the wind computation. The most crucial factor here is the "cruise speed" that is set on the "Aircraft" page (go to Menu/Aircraft) to open that page.

Please check and tell me if the speed you set on the "Aircraft" tab matches the high speed at which you are flying? Also, is the wind data set to realistic values (go to "Menu/Route and Wind" and switch to the 'wind' tab)?

Thanks again for your help,

Stefan.

fly-shadok, 2022-05-01:
Hi Stefan,

this would be an explanation, as I had not changed the default 'cruise speed' entry, which is still set at 150kph. I usually cruise much closer to that speed so I never noticed the inconsistency.

This is kinda unexpected behaviour though. I had expected the speed values would only be used for preflight planning, with inflight ETA/ETE being calculated from actual GS. Otherwise, any change in cruising speed or wind speed/directon would throw the prediction totally off. I would suggest changing the ETE/ETA calculation to being based on actual GS or adding a warning or explanation of this functionality (I didn't find one but also didn't really search, whoops).

Thanks for fixing item (2)!

kebekus, 2022-05-01:
> This is kinda unexpected behavior though. I had expected the speed values would only be used for preflight planning, with inflight ETA/ETE being calculated from actual GS. 

Agreed. I will try to improve the heuristics. It is of course always a little difficult to guess if the pilot deliberately chooses a different cruise speed, or changes the speed only briefly (e.g. for a quick descent). Also, I wonder what speed the app should assume for upcoming legs of the flight route…

kebekus, 2022-05-02:
I implemented an improved heuristic. If the bearing is in [-30°, 30°], then the app uses the (current ground speed)*cos(bearing) to estimate the speed for the current leg. For other legs, the wind-corrected "cruise speed" is used, as specified in the "Aircraft" settings.

The new heuristic will appear with the next major program version, which is most likely version 2.17.0.

@[fly-shadok](https://github.com/fly-shadok): Thank you for your help!

closed 2022-05-02 by kebekus