Akaflieg-Freiburg/enroute #235: new release not working with Lenovo tablet 2A 7-10F anymore

cessnapilot007, 2022-05-08:
I`ve used the program for a long time with my lenovo tablet TAB 2A 7-10F, android 5, with best success and istalled every single update.
With the new release it comes up for about 1,5 seconds, then shuts down.
I`reseted the tablet and installed the program several times, it does not run anymore.
Interesting is the icon on my tablet is now smaller then every other icon and also smaller as before.
Is any chance to get it fixed with a new release or should I look for another tablet?
Thanks an best regards,

kebekus, 2022-05-09:
@cessnapilot007 Thank you for the report. I have never seen this issue before, and I do not even have a guess of what is going on. That said, I can offer to look at the tablet if you are willing to send it to me by mail -- I would then set it to developer mode, so I can read out debug messages. On the other hand, Android 5 will not be supported for much longer. I have to update the underlying technology base later this year, which makes Android 7 the oldest supported version.

I am sorry that I have no better news.

cessnapilot007, 2022-05-10:
Thank you for your fast answer. I` ve installed the former version 2.16.2 which runs perfect again.
For testing purpose I`ve installed the 2.16.5. which does not run again, so I will stay with the 2.16.2.
For the future I will buy an Android 7 tablet.
Thanks and best regards from Munich.

kebekus, 2022-05-11:
Thank you for your report. Not knowing what else to do, I close this report now. Feel free to re-open if new information becomes available.

closed 2022-05-11 by kebekus