Akaflieg-Freiburg/enroute #237: lower airspace waypoints

joachimwe, 2022-05-15:
Hallo, ich nutze gerne die enroute App und möchte für Flüge mit flugplan gerne waypoints nutzen. Können z.b. enr lower airspace waypoints als weitere Kategorie neben Ad, nav und rep verfügbar gemacht werden? Danke, weiter so! Joachim

joachimwe, 2022-05-15:
E.g. https://opennav.com/waypoint/de 
Listen mit Koordinaten und country

kebekus, 2022-05-19:
Danke für den Vorschlag. Ich habe Pläne für eine Wegpunkt-Bibliothek, die die ENR Lower Airspace Waypoints enthalten könnte. Weil ich die App aber zurzeit allein programmiere, ist die Entwicklung sehr langsam. Ich hoffe, das Feature noch in diesem Jahr auf den Weg zu bringen, will aber lieber nichts versprechen. Vielleicht haben Sie Lust, mitzuprogrammieren?

Ich halte Sie über diesen Kanal auf dem Laufenden.

Grüße,

Stefan Kebekus.

kebekus, 2022-06-21:
Dear all,

Thank you for your suggestions, even if they took me a long time to implement. Version 2.18.0 of **Enroute Flight Navigation** will feature a "waypoint library." Users can add waypoints via the moving map or import a file in CUP, GeoJSON, or GPX format. 

Thanks again. I hope you are still enjoying the app.

Stefan.

closed 2022-06-21 by kebekus