Akaflieg-Freiburg/enroute #238: Custom base maps rendering issue (Beta)

manuuuu400, 2022-05-22:
Great to see the new feature in beta. Thanks a lot for your time and effort.

I noticed some quality issues when using my own raster mbtiles files.
You can see the example in Enroute:
[image]

And the correct rendering of the same file in OruxMaps:
[image]

So there seems to be a problem with the displayed resolution as well as some color shift.

Then I did some testing with a file from openmaptiles.org.
When zooming in, it seems that Enroute is switching to different zoom levels much later than other apps do.
Here is a screenshot of OruxMaps at the exact point it switches to the displayed zoom level (when zooming in)
[image]
However, to see the same zoom level in Enroute, I have to zoom in much much further (you can distinguish different zoom levels of openflightmaps due to different labeling):
[image]

This might be the cause for the resolution issue.
The color shift is also noticeable on the openflightmaps files (my second example), however I don't have an idea where this is coming from.

Just for some additional info:
I know that most raster maps don't look very sharp resolution wise on mobile devices, however you can greatly improve the quality when using larger image tiles (@2), which are also provided by openmaptiles for example.
So, when comparing the actual resolution of my provided screenshots you can see the difference of OruxMaps on the left, and Enroute on the right:
Normal resolution (@1):
[image]

Twice as many pixels (@2):
[image]

So the issue is not related to the higher pixel density of the device.

I hope this information helps.
Let me know if you need anything.

kebekus, 2022-05-22:
@manuuuu400 Thank you **very much** for this report. I will look into the issue.

kebekus, 2022-05-23:
@manuuuu400 Thanks again for your feedback. I would never catch these issues without help from our users. Today, I have uploaded program version 2.16.91 to Google Play, which hopefully fixes the issues that you describe (rendering quality, difference in color). Can you please try this out and let me know if things are working for you now? -- Best, Stefan.

kebekus, 2022-05-23:
@manuuuu400 Slightly unrelated: I guess you created the MBTILES, starting from GeoTIFF files? How did you do that? I googled around but did not find a satisfactory solution. Any hint is incredibly welcome!

manuuuu400, 2022-05-23:
@kebekus the color difference is gone and the quality seems much better.
There is still a difference of the displayed zoom level between Enroute and OruxMaps, although I don't know if this is just personal preference or some kind of a standard.
For me personally its good as it is now. I made a comparison of the difference (same as yesterday):
[image]
[image]

To your question about how to create mbtiles:
I found it most practical to use QGIS because it is completely free to use and is a very good allrounder.
You may find a lot of tutorials using MapTiler, although I wouldn't suggest it because the free version is not enough for the quality and size you need for a whole map.
In QGIS you can geo reference any file you have (JPG, PNG, PDF, etc.) and generate mbtiles of it. Geo referencing a large map needs some time and practice, but in the end you can get a perfect result.
If you are able to open a GeoTIFF file in QGIS (which I assume), then you can skip the geo referencing step and export whatever you have inside QGIS to mbtiles.

A lot of countries offer there VFR maps as PDF for free, which you can then convert by yourself.
When geo referencing, you can use the lat, lng grid on the map to set the reference points, which gives the best results.
Also a hint when using QGIS is to always use EPSG:3857. When you read the coordinates of the map (ex. 16.5 degrees) you can't enter 16.5 (this is EPSG:4326), you first have to convert it to EPSG:3857 (you can use this tool for example: https://epsg.io/transform#s_srs=4326&t_srs=3857&x=13.0000000&y=45.0000000 )

Feel free to contact me if you have further questions (German or English).

Btw. Enroute is the first free flying app (at least as far as I know) to support custom mbtiles. It will set this app apart from others when its out of beta ;)

kebekus, 2022-05-24:
The issue is solved in v2.16.91, closing this report now.

@manuuuu400 Thanks again for your help. Would you be willing to write a short explanation of how to use QGIS to convert GeoTIFF -> MBTILES? I guess many advanced users would find that helpful.

closed 2022-05-24 by kebekus

PilotNubilus, 2025-05-09:
@manuuuu400 not sure whether you are still around, but giving it a try anyway...

Does your described process of converting GeoTIFF to MBTILES in QGIS still work? If so, what QGIS Version are you using and how do you do it? I do exactly the same, but I have problems as described in my post #521.

Stefan kindly told me that the MBTILE files created by QGIS are broken. It seems legit, beacause the quality after importing the map is poor in a different app as well. He said that there are no tiles for some of the zoom levels, which are specified in the metadata to be included in the file. I cannot verify this myself, but the behaviour goes in that direction. When creating a map with only zoom level 12 selected, Enroute only shows it when zoomed in. When creating a map with alleged zoom levels 0 to 12, Enroute shows the map during all zoom levels, but with really bad resolution when zoomed out. It probably wants to display something, because it should be there, but can't because it isn't there and displays some garbage.

I looked a bit around in the GIS community and found a post from about two and a half years ago describing the same problem, but it was unanswered. There were some other posts as well, which did not quite match, but covered gdal not creating all zoom levels, while converting from TIF to MBTILES. The gdal library is used by QGIS, but it may or may not be relevant to this problem. And they were kind of beyond my IT abilities. I have just created a [post](https://gis.stackexchange.com/questions/492379/converting-mbtiles-from-tif-with-qgis-does-not-create-all-zoom-levels) there. Maybe someone there can help.

best regards

manuuuu400, 2025-05-11:
@PilotNubilus I am not creating my own maps anymore, I am getting along with the provided ones just fine.

After reading about your issue here and in your linked posts, it seems to be a QGIS related problem.
Since Stefan reported missing zoom levels above 8 after analyzing your file, that might be something to look into.
I remember some issues with missing low zoom levels caused by not completely covered tiles. So you could try to make the export coordinate rectangle smaller (or even add a second map behind yours, which contains the whole earth) to test if that's the case for you as well - chances are low however.

You could also try to uninstall QGIS and install an older version from the archive:
https://download.qgis.org/downloads/
Version 3.24.3 is what I have used back in the day.

PilotNubilus, 2025-05-13:
@manuuuu400 thank you very much for your response!

I have already tried using a smaller region, but did it again anyway, but it didn't help. Using a global map as background did not change anything either, but I liked that idea a lot. Furthermore I tried using many different export parameters in more or less a systematic way as well. When I read about using an old QGIS version I was absolutely sure, that this must work. Hence I was utterly surprised, that it did not. I am now confused and out of ideas. It not working with the old version points towards my process of doing it, or my computer, or my .tif file not beeing compatible with my workflow, but with Stefan's, or a combination of those. Of course it could also be something else. And it does not help, that I am wandering around in a field, where I don't know that much and have no one to ask for help.

To wrap things up, the problem probably lies somewhere in the triangle QGIS, myself and Enroute, where one or more of these factors can be the problem.

Thanks again for your time.
best regards