Akaflieg-Freiburg/enroute #240: Latest enroute version on flathub or snap compiled for arm

christianhbeck, 2022-05-30:
Hello,

I would like to ask for getting regular updates for the linux versions compiled for arm. snap version is 2.6.5 and flat hub 2.15.1 (arm). 

I have started running enroute directly on the raspberry pi4 I use in my stratux device to which I have attached a touch screen display. Both existing options for installing (flat hub and snap) work on this platform.

A very nice additional feature would be to have a qt based virtual keyboard appear when there is text or number input required by the app. I am not sure if this feature is actually present and is removed by a flag during compilation for desktop. I have not found practical virtual keyboard solution at the OS level.

regards,
Christian

kebekus, 2022-05-31:
Dear Christian,

Thank you for your request. I understand what you are asking, but I am afraid I cannot deliver. I develop the app alone, and in my spare time and already now, maintenance of the app and its infrastructure takes more time than I can reasonably invest. I am therefore closing this issue now.

However, the app is 100% Open Source, and everybody is welcome to join! Please let me know if you are interested in maintaining binary releases for the ARM platform. I will be more than happy to help and get you started. Just comment in this thread. Even though the issue is closed, GitHub will notify me automatically.

Best,

Stefan.

closed 2022-05-31 by kebekus

christianhbeck, 2022-06-11:
Hi Stefan,

I am willing to give it a try. My guess is that the steps are:
1. figure out how to build the binaries
2. push potential modifications to the git
3. be able to push the binaries to repositories. I am not sure if you want to delegate this

I tried step 1 and failed. 
I plan to build on the target platform as a first step. I added the following packages to the system: 
qtbase5-dev qtchooser qt5-qmake qtbase5-dev-tools qttools5-dev qtpositioning5-dev libqt5svg5-dev clang llvm ninja-build

when I launch the build script I get the output below. The cake warnings suggest that there are missing make files? Can you give me a hint?

regards,
Christian

pi@stratux:~/enroute-2.16.5 $ ./buildscript-linux-debug.sh 
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Found Git: /usr/bin/git (found version "2.30.2") 
Not working from a GIT directory
CMake Error at 3rdParty/CMakeLists.txt:13 (add_subdirectory):
  The source directory

    /home/pi/enroute-2.16.5/3rdParty/qhttpengine

  does not contain a CMakeLists.txt file.

CMake Error at 3rdParty/CMakeLists.txt:36 (add_subdirectory):
  The source directory

    /home/pi/enroute-2.16.5/3rdParty/KDSingleApplication

  does not contain a CMakeLists.txt file.

CMake Error at 3rdParty/CMakeLists.txt:43 (add_subdirectory):
  The source directory

    /home/pi/enroute-2.16.5/3rdParty/sunset

  does not contain a CMakeLists.txt file.

CMake Error: File /home/pi/enroute-2.16.5/3rdParty/enrouteText/desktop/linux/de.akaflieg_freiburg.enroute.appdata.xml.in does not exist.
CMake Error at metadata/CMakeLists.txt:6 (configure_file):
  configure_file Problem configuring file

CMake Error: File /home/pi/enroute-2.16.5/3rdParty/enrouteText/desktop/linux/de.akaflieg_freiburg.enroute.desktop.in does not exist.
CMake Error at metadata/CMakeLists.txt:7 (configure_file):
  configure_file Problem configuring file

-- Configuring incomplete, errors occurred!
See also "/home/pi/enroute-2.16.5/build-linux-debug/CMakeFiles/CMakeOutput.log".
ninja: error: loading 'build.ninja': No such file or directory