Akaflieg-Freiburg/enroute #241: MBtiles as background for the vector maps.

tue-kyndal, 2022-06-06:
Hi Stefan.

Lovely to see the added support for MBTILES. 

Grat job ;O)

I have generated a set of MBTILES for Copenhagen area 41 edition, with zoom level 1-14 96dpi.
Based on the official version from 
https://aim.naviair.dk/media/1155/edition-41-icao-anc-1250000-copenhagen-area-geo-tiff-300dpi.tif

It can be downloaded here:
http://tue.kyndal.dk/kort/edition-41-icao-anc-1250000-copenhagen-area-zoom-1-14.mbtiles 

It loads just fine, but it overrides the vector data, so you can not pick WPT etc. any more.
This is bit of a draw back.

Is it possible just to load the raster and only set it as a background, behind the vector layer?
This way you would have the best of both worlds, vector graphics and way points etc,  and useful map info , city names, lakes, roads etc from the official ICAO map.

Also, you can only import using a third app, and selection open with Enroute, and the only way to disable i is to fully uninstall the map.
It could be great if you could select the map file from within the Enroute app, and if it could be enabled/disabled in the Base Map overview, as required for the flight.

Keep up the good work.

I'm already trying to spread the work about this great app and new feature in the danish flying community ;O)

[image]

kebekus, 2022-06-08:
@tue-kyndal Thank you for this nice message. The feature that you request is already present in part. You need to install the aviation map together with the raster map. The aviation data is then not drawn but is considered: if you double-tap on an airfield, then the airfield data will be shown; ditto for airspace data. So yes, you can pick waypoints.

I experimented a little with drawing vector and raster data. The result is often an unreadable mess. I also experimented with options to enable/disable maps on the fly, but I found that this makes the app too complicated -- simplicity is the app's main selling point, so I try to avoid options as much as I can.

kebekus, 2022-06-08:
I am closing this issue now because there is no further action for me to pursue. Thanks again!

closed 2022-06-08 by kebekus