Akaflieg-Freiburg/enroute #242: Installing on Raspberry pi 4 64 bit,  Raspberry OS

ocean71, 2022-06-06:
I am having difficulty running the app on a Raspberry Pi. It has compiled and built OK, but when I come to run the app it loads up Enroute Flight Navigation but with a blank screen and no way to access anything. The following information is in the terminal when I run ./enroute:

QObject::connect: No such signal QDBusAbstractInterface::ActionInvoked(uint,QString)
QObject::connect: No such signal QDBusAbstractInterface::NotificationClosed(uint,uint)
QQmlComponent: Component is not ready
qrc:/qml/main.qml:493:5: QML StackView: initialItem: qrc:/qml/pages/MapPage.qml:22 module "QtPositioning" is not installed
qrc:/qml/pages/MapPage.qml:21 module "QtLocation" is not installed
qrc:/qml/pages/MapPage.qml:22 module "QtPositioning" is not installed
qrc:/qml/pages/MapPage.qml:21 module "QtLocation" is not installed
qt.qpa.xcb: QXcbConnection: XCB error: 5 (BadAtom), sequence: 471, resource id: 0, major code: 20 (GetProperty), minor code: 0

I am clearly doing something wrong, but I am not sure what. I got no error messages when compiling or building.

Would be very grateful for any help.
Many thanks.

kebekus, 2022-06-08:
Dear ocean71,

I am unsure how much I can help. Looking at the debug messages, it does seem to me that several components of Qt (namely: QrLocation, QtPositioning, and perhaps DBus) are unavailable on the machine where you run the app. Perhaps they were not installed properly, or are not part of a standard Qt installation on the Raspi?

I am not sure there is more that I can say…

Best,

Stefan.

kebekus, 2022-06-08:
I am closing this issue because I am unsure what to do. Please feel free to contact me by e-mail if there are specific questions that I might be able to answer.

closed 2022-06-08 by kebekus