Akaflieg-Freiburg/enroute #245: Show mountain peak elevation in map

kebekus, 2022-06-12:


think-nice-things, 2022-09-10:
Hi,

I don't know how you want to show elevation but here's a suggestion.
Since dem / hgt data are installable, it should be possible to highlight regions which are above the current in-flight altitude.
Here's an example (not from a flight app but taken from a flight planning with QMapShack where I set the altitude to 11700 ft):

[image]

kebekus, 2022-09-12:
@think-nice-things Very nice flight track. I was flying in that region with a glider this summer.

kebekus, 2022-09-12:
@think-nice-things Welcome back! If you are ever interested in helping with the programming, you're most welcome!

kebekus, 2022-09-12:
@think-nice-things As for your suggestion: I had something much easier in mind. At present, the moving map of Enroute Flight Navigation shows a few selected mountain peak names ("Matterhorn"). All I want is to add the height ("Matterhorn 4478m").
The difficulty is to make the map switch units, depending on the user preference ("Meters" or "Feet").

Your suggestion makes perfect sense to me, but seems impossible to implement efficiently with the map display technology (MapBox GL) that I currently use. If you have a suggestion for an implementation, let me know.