Akaflieg-Freiburg/enroute #247: GPS-Empfänger

Petersen78, 2022-06-17:
Trotz aktivierter Standortbestimmung in ENR FLIGHT INFORMATION APP erzeugt die App die Fehlermeldung, dass keine GPS-Positionsdaten empfangen werden

kebekus, 2022-06-22:
Lieber Petersen78,

entschuldigen Sie die späte Antwort -- ich bin zurzeit beruflich auf Reisen,

Ich helfe Ihnen gern, benötige dazu aber viel mehr Informationen.

- Welche Version der App verwenden Sie?
- Welches Betriebssystem verwenden Sie? Desktop? Android?
- Können Sie den Fehler genauer beschreiben? Vielleicht mit einem Screenshot oder einem Screencast Video?
- Funktionieren gleichzeitig andere Apps mit Positionsbestimmung?
- Sind Sie im Freien? Die App verwendet nur Satellitennavigation, die in Innenräumen meist nicht geht.
- Was wird angezeigt, wenn Sie im Hauptmenü die Seite "Info/Positionsbestimmung" wählen? Auch hier ist vielleicht ein Screenshot hilfreich.

Beste Grüße,

Stefan Kebekus.

kebekus, 2022-06-29:
Ich schließe diesen Vorgang, weil ich noch keine Antwort bekommen habe. Falls Sie noch antworten, öffne ich den Vorgang gern wieder.

closed 2022-06-29 by kebekus

Petersen78, 2022-10-11:
Lieber Herr Stefan Kebekus, die Antwort haben Sie mir jetzt schon gegeben,  ich habe die App in einem Innenraum ausprobiert und dann logischerweise keine Positionsdaten empfangen.  Im Freien , wo es einzig gehen muss,  geht es natürlich...Vielen Dank, Peter Petersen --Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.