Akaflieg-Freiburg/enroute #250: Keep current name of route when displaying rename dialog

rawtaz, 2022-06-28:
**Is your feature request related to a problem? Please describe.**
When in the library you click to rename a route, you currently have to type the entire name all over again. This is very cumbersome when you just want to change part of the name.

**Describe the solution you'd like**
Make the old/current name of the route be the prepopulated value when displaying the rename dialog, such that one can just edit the current name.

**Describe alternatives you've considered**
I tried copying the old name but there's no way to do that it seems.

**Additional context**
None.

kebekus, 2022-06-29:
Thank you for this report. I will look into this and implement your suggestion ASAP.

kebekus, 2022-06-30:
Thanks again. The suggestion is implemented in version 2.18.1, which I uploaded to Google Play a few minutes ago. It should become available withing the next 24h.

Best wishes,

Stefan.

closed 2022-06-30 by kebekus