Akaflieg-Freiburg/enroute #251: Enroute fot Hot Air Ballooning

Balloondriver, 2022-07-03:
Hello 
I am using enroute for Hot Air Ballooning.  It is very good. But the one thing which
 I miss is a menu feature, that allows to adjust the projection line of the aircraft symbol. It is fixed to show the track 5 minutes ahead. While this might be good for a glider, a hotAir balloon moves much slower. So it would be desirable to have this showing the present track as an indefinite line. Or if that is not possible for an hour ahead, instead of 5 minutes. That would make it a lot more easy to see where the Balloon is drifting to.

Thanks and kind regards

kebekus, 2022-07-05:
Thank you for your suggestion. I never expected that balloonists would use this app. I understand your point and will try to find a solution. However, please do not expect me to be extremely fast. I program the app alone, and in my spare time, so development is relatively slow. I shall keep you updated via this channel.

Best wishes -- Stefan.

Balloondriver, 2022-07-05:
Hallo Stefan

Darf ich in Deutsch schreiben?
Vielen Dank für deine rasche Antwort.
Ja auch wir Ballonfahrer müssen navigieren...
Kenne jetzt doch einige Kollegen welche die App dafür nutzen. Ich muss
sagen, enroute ist für mich etwas vom besten was ich bis jetzt gesehen
habe. Und das ohne viel unnötigen Ballast.

Freundliche Grüße
Peter / HB-QOO

kebekus, 2022-07-05:
Danke für das Lob! Schaun' wir mal, was ich machen kann…

tue-kyndal, 2022-07-05:
It's not a bad idea to add an option menu for the projection line length. 

If you could define the length of the line, I would use 6 minutes as standard. 

In this case the length of the line would be 0.1 of your indicated airspeed in the same unites. 

It would be easy to give a time and distance ATC for a upcoming waypoint. If you are doing 110kn, then the distance is 11miles and 6 minutes when the aero hits the mark. 

Just a thought. 

NB. I did a balloon flight with a Danish balloon captain. Their Ballon association have developed a navigation app for iPad, that show exactly what you are asking for, and some extra Ballon specific information etc. 

Perhaps contact some Danish balloon captains for information.

kebekus, 2024-01-23:
Ein Ballonfahrer schreibt:

1. Gibt es eine Möglichkeit die Flug/Fahrtprojektionslinie auch auf mehr als 5 Minuten einzustellen? Ich verstehe das für die Fliegerei bei deutlich höherem Speed die 5 Minuten aussreichend sind, bei uns Ballonfahrern sind mit den langsameren Geschwindikeiten 5 Minuten sehr wenig.
2. Die Annäherung an einen Routen bzw. Wegpunkt hört bereits in einem (für uns) relativen großen Radius auf, gäbe es auch hier eine Möglichkeit das Ganze etwas enger auf den jeweiligen Punkt zu navigieren?