Akaflieg-Freiburg/enroute #252: Enroute build for F-Droid

ninelima, 2022-07-09:
Hello Stefan,  

I am attempting to create a build of Enroute version 2.18.1 for FDroid. But I run into this problem.  
It appears that "de.akaflieg_freiburg.enroute/3rdParty/android_openssl/latest/arm/libcrypto_1_1.so" does not build for me.  
  
I have tried numerous things without success. It is currently beyond my skill level to resolve this issue.  

For the complete build log,  see: https://gitlab.com/ninelima/fdroiddata/-/jobs/2697597442  
For the FDroid MR, see:  https://gitlab.com/fdroid/fdroiddata/-/merge_requests/11319  
  
Here is an extract when the build fails:  
  
```
[100%] Built target enroute
+ /builds/ninelima/fdroiddata/build/Qt/5.15.2/android/bin/androiddeployqt --input android_deployment_settings.json --output android-build --release --apk de.akaflieg_freiburg.enroute_290.apk
Generating Android Package
  Input file: android_deployment_settings.json
  Output directory: /builds/ninelima/fdroiddata/build/de.akaflieg_freiburg.enroute/build-android-release/android-build/
  Application binary: enroute
  Android build platform: android-29
  Install to device: No
Warning: QML import could not be resolved in any of the import paths: enroute
External library /builds/ninelima/fdroiddata/build/de.akaflieg_freiburg.enroute/3rdParty/android_openssl/latest/arm/libcrypto_1_1.so does not exist!
  -- Skipping /builds/ninelima/fdroiddata/build/Qt/5.15.2/android/plugins/position/libplugins_position_qtposition_serialnmea_arm64-v8a.so. It has unmet dependencies: lib/libQt5SerialPort_arm64-v8a.so.
==== detail end ====
```

Any suggestions would be appreciated.
Thank you

kebekus, 2022-07-11:
@ninelima I received your message, but I will need a little time to answer. Sorry for the delay.

ninelima, 2022-07-20:
The F-Droid build recipe is quite involved. It was originally done by other packagers and I have been tyring to get a build working for the latest versions.

It is fairly close (I think) but there are issues with building libcrypto_1_1.so which seems to be a dependancy.

See: https://gitlab.com/ninelima/fdroiddata/-/jobs/2697597442

I am a bit stuck here :-|

kebekus, 2022-11-30:
I contacted the person who uploaded the app to F-Droid a while ago and asked if he is willing to help. Let's see if he answers.

Best,

Stefan.

kebekus, 2022-12-02:
@ninelima Georg Küngemann, who uploaded the app to F-Droid a while ago, has replied. He promised to update the app on F-Droid. He would also be glad to answer your questions. Perhaps the two of you can team up, in order to ensure that the app on F-Droid remains up-to-date?

I am hesitant to post Georg's e-mail address on this public server. Could you please send me an e-mail privately, so I can get the two of you in touch? You can find my address in the app.

kebekus, 2022-12-14:
@ninelima Georg Küngemann has compiled the latest version of the app for F-Droid. I attach a translation of an e-mail that he sent me last night. The e-mail also contains a link to GitLab, where he hosts the recipie that he uses to compile the app.

@ninelima I guess this answers your question, so I will close this issue now. Please feel free to re-open it if appropriate.

--

> it finally worked to build Enroute Flight Navigation again with the F-Droid build process. In the yml file for the F-Droid build process, I included the build for version 2.21.3 and adjusted it a bit. The build process runs through again with this:
https://gitlab.com/georgvfr/fdroiddata/-/jobs/3469518588

> I still need to test the finished apk file on the tablet. After that, I would make a merge request to add the new version to the F-Droid data repository. This would then make the new version available again via F-Droid.

closed 2022-12-14 by kebekus

GeorgVFR, 2022-12-14:
Hello,
the updated app was tested successfully on my tablet and the merge request to update Enroute Flight Navigation was created on F-Droid's data repository:

https://gitlab.com/fdroid/fdroiddata/-/merge_requests/12264

Best regards,
Georg

kebekus, 2022-12-15:
@GeorgVFR Thank you for that. I greatly appreciate!

rawtaz, 2023-04-17:
@GeorgVFR I see you've done some seriously great work on this, but that it's now stuck on the F-droid side of things. Is there any way you could poke the project on that side (https://gitlab.com/fdroid/fdroiddata/-/merge_requests/12264) to see if we can get some traction? Flying season is now here, it seems :-)

kebekus, 2023-04-26:
Dear All, 

I had a (very) brief look at the issue; please take a look at a few notes below. Much as I would like to help, my job has become quite stressful recently, and I regret that I do not have the time and capacity to delve into the matter more seriously. 

Best,

Stefan.

--

- Are you trying to build the app against Qt5? This will no longer work.
- The app needs to be built against Qt 6.4.3 exactly.
- Qt 6.4 does presently not come with the QtLocation module. Qt6.5 has a QtLocation module, but the MapLibre renderer is presently missing. As a result, the QtLocation module and the renderer need to be installed/compiled manually
- One solution is to the use pre-built binaries from this site: https://github.com/maplibre/maplibre-native-qt/releases/tag/v2.1.0
- Another solution (which I presently use for the Android binaries) is to compile everything myself, from this repository: https://github.com/Akaflieg-Freiburg/qtlocationstatic

If you want to compile everything yourself, you can have a look at the scripts that I use in GitHub action to do the compilation:

- https://github.com/Akaflieg-Freiburg/enroute/blob/main/.github/workflows/android.yml

I hope that building Enroute will become easier in the future, when decent modules for Qt6.5 become available.