Akaflieg-Freiburg/enroute #253: Custom ANSPs based maps

ElSabio97, 2022-07-12:
**Describe the solution you'd like**
I would like to have an option in the library for maps designed by ANSPs (Air Navigation Service Providers) like the ones made by ENAIRE for VFR flights (LE1 to LE6): https://aip.enaire.es/AIP/CartasInsigniaImpresas-es.html

tue-kyndal, 2022-07-13:
Just load the map in Qgis and export it as a MBTILES map. The you can load it as custom map. Load the vector data maps afterwards. Then you also have an active vector layer for wpt etc. 

If the map is all ready orientated it's very easy. If it's not you need to Geo reference the map first.

There are tons of guides online how to do this, im sure you can find on en fransh as well.

kebekus, 2022-07-15:
@ElSabio97 Thank you for the report, I appreciate. As @tue-kyndal pointed out, it is possible to use maps ANSP maps in Enroute Flight Navigation. However, the process requires the use of an external program (like QGIS) that transforms the maps raster maps to the pre-rendered MBTILES format that Enroute uses for performance reasons. [The manual explains the process in detail.](https://akaflieg-freiburg.github.io/enrouteText/manual/02-advanced/customMaps.html#raster-maps-in-geotiff-format)

I do not see how this can be automated: ANSP web sites change all the time, and someone would have to check them regularly for updates. Perhaps worse, if we offer the converted files to our users, we will definitively run into copyright issues. Contracts would have to be negotiated with all relevant ANSPs. Also note that conversions are CPU-intense, so someone would have to set up (and finance) servers that do this job.

I am closing this issue because I do not see an easy solution. If you are willing to do the necessary work, then please feel free to contact me, and I be more than glad to assist.

Best wishes,

Stefan.

closed 2022-07-15 by kebekus