Akaflieg-Freiburg/enroute #254: Clarify F-Droid relation and maintenance

rawtaz, 2022-07-13:
Sorry, but this isn't a regular feature request, but a place for a quick discussion to clarify the F-Droid distribution of Enroute. I opened a new issue instead of hijacking #252 because even if it's on the same topic it's not really the same thing. So please consider this just a place to record the result of the following discussion to make it searchable - once we've settled on a clarification we can close this :)

I'm using the F-droid version of Enroute, which by now is quite old and doesn't have the most recent enhancements. I got curious who manages the releases of the F-Droid version, but all I could find in F-Droid was that the app is by "Stefan Kebekus (not F-Droid maintainer)". I did see that @ninelima has been working on this earlier, but it's not clear to me if they're the one who manages the F-Droid distribution.

So in short, who publishes the app on F-Droid? And what is needed to make new releases automatically show up on F-Droid (once #252 is dealt with, of course)?

kebekus, 2022-07-19:
Message has arrived, but I am under water with my job and need a few days to answer.

rawtaz, 2022-07-19:
No worries, don't stress yourself out, you have a life that has priority over Enroute.

I was hoping maybe @ninelima could have some insight, seeing as they seem to have helped out with the F-Droid distribution previously :)

ninelima, 2022-07-20:
@rawtaz 
See: https://github.com/Akaflieg-Freiburg/enroute/issues/252#issuecomment-1189728708
for the latest developments

kebekus, 2022-11-06:
Dear all,

sorry that I have left this report unanswered for a very long time. Maybe that isn't relevant anymore, but let me answer anyway.

I have nothing to do with the program version that exists at F-Droid. The binary has been compiled and uploaded by someone else, and I do not even have an account (or whatever is needed) with F-Droid. Also, I have never tried the program version there.

In principle, I very much support the idea for F-Droid as a non-commercial source for free software. I have therefore added a sentence at the [Enroute home page](https://akaflieg-freiburg.github.io/enroute/download/#unofficial-versions): "An unofficial version of the app is also available at F-Droid. We endorse publication at F-Droid, but we have not tested this unofficial app for quality." I will update that sentence shortly.

I am closing the issue now because I do not see what else to do. Please come back to me anytime. I am glad to help, even though I have to admit that my time is severely limited.

Best,

Stefan.

closed 2022-11-06 by kebekus