Akaflieg-Freiburg/enroute #262: Airspace entry alarms

fionn1989, 2022-09-04:
I would suggest to include an airspace entry alarm system which alerts the pilot in case he is invading P, D, R or any controlled airspace. 

Very thanks for your wonderful work!

anthraxn8b, 2022-09-04:
Great idea! Maybe configurable to be alerted Xmin before to have enough time to communicate.

kebekus, 2022-09-07:
@fionn1989 @anthraxn8b Thank you for your suggestions! These features have been on my TODO list for long. However, given the current speed of development, I do not expect that anything will be implemented before spring 2037 :-( I am doing the programming alone and in my spare time…

If you are interested in helping with the development (or if you know anyone who does), this would be highly welcome. The main issue with "Airspace entry alarms" is Android background processing -- the alarm needs to work even if the display is switched off. Because of privacy concerny, background processing is highly regulated on Android, and not easy to use. Anyone with knowledge of the platform is highly welcome.

kebekus, 2022-09-08:
@anthraxn8b You wouldn't believe it. Over time, I received way more than 1k requests for features, and I consistently ask if people are willing to help. Result: zero. And yes, the Akaflieg Freiburg supports the project. Still I am the only one doing the programming.

Any help is greatly appreciated.

fionn1989, 2022-09-08:
I mostly develop in Python in my job, but let me have a look at your project structure to try to understand your arquitecture and technologies you are using (I.e how do you inject a C++ app into Android VM). I used to code C and java several years ago, so maybe I could help a little bit.

kebekus, 2022-09-09:
@fionn1989 Thank you for your reply. This is very encouraging news. I will come back to you next week, when I return from vacation.