Akaflieg-Freiburg/enroute #263: can't get custom mbtiles map working

think-nice-things, 2022-09-10:
I tried custom mbtiles map which should be possible according to the manual.
I tried different mbtiles maps, both created myself (with mobac) and also some which are publicly available (e.g. the ones from openflightmaps).
All the maps are working with different map applications as oruxmaps, so the maps should be ok.
Sharing with enroute seems to work, when sharing enroute prompts me for a map name, afterwards the imported map appears under background raster maps and all other raster maps are uninstalled.

But: after having installed the custom map, the map screen is just empty.
I tried to zoom in or out or pan the map but it stays empty.

I tried with two different android devices, the problem is the same for both devices.

this is Enroute 2.20.3 as from the google play store.
Devices: Xiaomi Mi A1 and Mi Pad 4, both with custom roms.

**Any ideas?
Can anyone provide a custom map which is supposed to work?**

[image]

kebekus, 2022-09-12:
@think-nice-things Thank you for the report. The easiest way to get a working MBTILES file is to download one from open flightmaps, for instance here:

- https://www.openflightmaps.org/ed-germany/

If you want, I can send you a file that is guaranteed to work on my machine. Also, if you send me your file, I could take a quick look and check if I can spot the issue immediately.

kebekus, 2022-09-18:
@think-nice-things Any news here?

think-nice-things, 2022-09-18:
no, unfortunately not. The mbtiles from openflightmaps do not work as well as my self generated mbtiles.
The ed_256.mbtiles as from openflightmaps seems to be correctly imported and show up in the map dialog but the map view is empty like in my first post.
[image]

tue-kyndal, 2022-09-18:
Just use Qgis to generate the MBTILES and do as the guide instructs.
It works just fine.
If the map is not properly geo referenced it will not work. Obviously. 
You can do this your self using the lat log indications on the edge of a VFR map, before generating the MBTiles.
You get a nice report of the accuracy you achieved.
Check my maps here
[
](https://drive.google.com/drive/folders/1r6j_2JCkEMIvKj3FkF--2K5h4ockGKw2?usp=sharing)

think-nice-things, 2022-09-18:
@tue-kyndal your files were helpful. They contain in metadata "minzoom" and "maxzoom" entry which are not present neither in my self generated mbtiles maps with mobac nor in the ones from openflightmaps.
If I add minzoom/maxzoom metadata "manually" to openflightmaps maps or self-generated mbtiles they show up perfectly in enroute.

[image]

tue-kyndal, 2022-09-18:
Correct.

The zoom levels must be set.

I also realized this.

After some experiments I figured 1-12 seem to be the best concidering the file size etc.

Let me know what zoom levels you find bedst.

Nice to be able to help 🤠

think-nice-things, 2022-09-18:
Hmm. Your files claims version 1.1 but contains minzoom and maxzoom. According to https://github.com/mapbox/mbtiles-spec minzoom and maxzoom appeared in mbtiles version 1.3.

According to [1.3](https://github.com/mapbox/mbtiles-spec/blob/master/1.3/spec.md) minzoom and maxzoon is recommended but not mandatory. In the last section ["Future directions"](https://github.com/mapbox/mbtiles-spec/blob/master/1.3/spec.md#future-directions) they write "In a future revision of this specification, the bounds, minzoom, and maxzoom rows of the metadata table will be mandatory."

It is easy now for me to modify my mbtiles files since I know I've to add minzoom and maxzoom. BUT: all my existing maps I'm using with other apps -- and most promiently those from openflightmaps and avioportolano -- do _not_ contain minzoom and maxzoom. I wonder if it would be helpful if enroute could handle those.

kebekus, 2022-09-18:
Your reports are extremely helpful! I understand the issue now and hope to have a bugfix out fairly soon.

@think-nice-things @tue-kyndal Live would be easy if all bug reporters were as competent as you are. Let me know if you interested in joining the development!

kebekus, 2022-09-19:
Dear all, 

I just pushed a new version (=2.20.5) to the beta channel of Google play. Can you please check and confirm that this solves the issue? -- Thanks!

Best,

Stefan.

kebekus, 2022-09-27:
I assume that the issue has been resolved in 2.20.5.

Thank you for your help!

closed 2022-09-27 by kebekus