Akaflieg-Freiburg/enroute #266: idea | attach notes to waypoints

anthraxn8b, 2022-11-29:
Adding a note to a waypoint combined with an alert maybe helpful (climb/decent now,...).

kebekus, 2022-12-12:
@anthraxn8b Thank you for the suggestion. I will try to implement waypoint notes during the holiday break (that is, if my family lets me…).

Best,

Stefan.

kebekus, 2023-01-01:
@anthraxn8b "Notes for waypoints" are implemented and will appear with the next release. I expect a longer beta-phase than usual, because of a major tech update in the underlying libraries.

Waypoint alerts are on my TODO list but will not appear soon. The problem is that the app needs to monitor the position even when the device display is off and/or other apps are in the foreground. Google's Android policies for background processing are a nightmare, and many vendors force-install crappy "battery saving apps" that randomly kill background processes that are deemed to use too much energy. Working around all this looks very much like a nightmare…

I close this issue now. Please feel free to re-open.

THANK YOU FOR YOUR HELP -- HAPPY NEW YEAR!

closed 2023-01-01 by kebekus

anthraxn8b, 2023-01-02:
Is there a description how Google Maps handles that?
On my phone (Pixel4a) Google Maps runs PiP if in the background.
Maybe they would offer help for a Akaflieg project...

kebekus, 2023-01-02:
@anthraxn8b Yes.

- [This document](https://developer.android.com/guide/background) describes Android background processing.
- [This website](https://dontkillmyapp.com/) explains the problems with crappy battery saving software.

As you will see, the battery saving software often makes extra exceptions for big-vendor software (Google, Facebook, …) while acting pretty randomly on generic programs. There are of course ways to work around this, but that will take some quality time, which I do not really have at present. As a result, development is slow…

I have repeatedly asked the community for help, but so far no-one has volunteered. Are you interested in joining the development? Any help is greatly appreciated!