Akaflieg-Freiburg/enroute #267: Data feature

2dylan1, 2022-11-30:
Sometime I fly with my tablet and sometime with my phone. The problem is that the waypoint I always ad on manually.
I would like to have the app make it possible to have all my waypoints and reporting points on all my devices.
I do not have time to add all the waypoints to my different devices.
Maybe make it possible to sign in with a username and password then let all data link to all devices. 
Another request, to record flight data like routing, altitude and groundspeed so one can review it afterwards. 
If you look at Easy plan it has that feature.

kebekus, 2022-12-02:
@2dylan1 Thank you for your suggestions. The feature that you request ("sharing waypoints") is already partially implemented: have a look at the section "Export and share the waypoint library" in the manual, where it explains how to transfer the waypoint library between devices:

- https://akaflieg-freiburg.github.io/enrouteText/manual/02-steps/waypointLibrary.html

I understand that this is not 100% the solution that you were asking for. That, however, will not appear quickly. I am developing this free app alone and in my spare time. As long as no-one helps with the programming, setting up (and maintaining) new infrastructure is simply beyond my capacities. It is also unclear who is going to pay for the infrastructure.

kebekus, 2022-12-02:
@2dylan1 Flight data recording has already been requested:

- https://github.com/Akaflieg-Freiburg/enroute/issues/44

Again this feature is not going to appear quickly because no-one is currently helping with the development.

kebekus, 2022-12-02:
@2dylan1 Please do not be upset, but I am closing this request for the reasons explained above. I will re-open the issue in case that there will be some progress towards implementation.

You might want to go to

- https://github.com/Akaflieg-Freiburg/enroute/issues/44

and subscribe to the issue there. This way, you will be informed automatically as soon as any progress is made.

Best,

Stefan.

kebekus, 2022-12-02:
@2dylan1 Ah, yes, and then 

- Thank you for your contributions!
- Please keep reporting any bug that you might find! Fixing bugs is my top priority.

closed 2022-12-02 by kebekus