Akaflieg-Freiburg/enroute #269: Warn if app won't be useful in your country

jidanni, 2022-12-31:
Perhaps on the first page of the app, and also in the library when we go to download maps, and also on Google Play, warn that the app probably won't be useful if one's country hasn't been added to the list of downloadable items. That way users wouldn't install the app to only find out that they can't use it in their country.

And maybe have a list of countries that it's currently usable in right there on Google Play.

kebekus, 2022-12-31:
@jidanni Thank you for your input. I am unsure about Google Play, but a warning on first startup definitively useful. I expect to implement your suggestion in Q1 of 2023 … but since I program the app alone, development is slow and I cannot make any promises.

Just for curiosity: where are you flying?

jidanni, 2023-01-01:
Thanks!
Taiwan.

kebekus, 2023-02-17:
@jidanni Thanks again! The feature you asked for has been implemented and will be in the next major version.

closed 2023-02-17 by kebekus