Akaflieg-Freiburg/enroute #270: Not working app

2dylan1, 2023-01-18:
My app doesn't want to work
[image]

kebekus, 2023-01-18:
@2dylan1 Thank you for your report! I will investigate the issue and try to come up with a solution as soon as possible. Can you please help me with the following questions:

- What device do you use? Is this a standard phone? Do you know the maker and/or the device name (such as "Samsung Galaxy A7")?
- In your screenshot, it seems that the app has started, but the moving map screen stays blank. Does the rest of the app work besides showing no map? Does the menu open when you tap on the button on the top left? Can you open the "nearby" page?
- What version of the app do you use? The version number is shown on top of the drawer that opens when you click the menu button.
- Do you know what version of the Android system your device uses?

Best wishes,

Stefan.

2dylan1, 2023-01-18:
Hey Stefan

All the other functions works on the app it's only the map not working. I
am able to enter the nearby page and checkpoints, everything. It never had
a problem on my cellphone before. I am using a Xiaomi Redmi note 11pro, and
the app version is "v 2.22.18"

If you can aid me in the problem, I will highly appreciate it.

Kind regards
Dylan Smith
+27 72 373 4754

2dylan1, 2023-01-19:
> @2dylan1 <https://github.com/2dylan1> Thanks again. The app's underlying
> technology has recently seen an upgrade; this seems to cause problems on
> your device. Since you have kindly agreed to take part in the beta-testing
> of this app, you see the new version earlier than others. This explains why
> this is the first time you have experienced the problem.
>
> The app cannot load the plugin required to draw the map. I am currently
> trying to understand why this is the case. The problem does not show on any
> of my machines, which makes things extra difficult for me. The log files of
> your device might hold the clue. Reading log files from your device is,
> however, complex. How well do you know the Android system? Are you an
> Android developer yourself?
>
> Best,
>
> Stefan.

Hi Stefan

I'm not that clued up with the system because I am only now using it for a
while. I have a friend that can maybe check it out for me. Please let me
know if I can assist in any way

kebekus, 2023-01-20:
@2dylan1 Thanks for offering to help. If your friend knows how to use the "Android Developer Bridge (=ADB)," that would be helpful. I am preparing a version of the app that gives more precise system information, helping me identify the problem. I will return to you as soon as the new version is ready.

2dylan1, 2023-01-20:
Thanks for the update. I will see what I can do and organise. Thanks

kebekus, 2023-01-25:
The problem has resolved itself. Thanks again for the report.

closed 2023-01-25 by kebekus