Akaflieg-Freiburg/enroute #271: Git recursive clone broken

betzuka, 2023-01-22:
I think there is a problem with .gitmodules, specifically KDSingleApplication, this fails:

```
Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:Akaflieg-Freiburg/KDSingleApplication.git'
```

kebekus, 2023-01-22:
Understood. It works for me because I use SSH keys to access GitHub. I will correct it ASAP, but probably not tonight (the kid needs to go to bed…)

kebekus, 2023-01-23:
@betzuka Thanks again for your report. I pushed an update to the master branch just now. Can you confirm that checking out submodules works now for you?

If you try to compile, the next obstacle is probably the missing dependency "QtLocation." I'll be writing about that later. I will try to do that tonight, but I cannot promise.

betzuka, 2023-01-23:
> I pushed an update to the master branch just now.

I don't think you pushed it.

kebekus, 2023-01-23:
@betzuka You're right that I forgot to push. I did that now. The path to enrouteText is unchanged because some users need to write to GitHub via that channel. If you can confirm that things work now, then I will close this issue.

kebekus, 2023-01-24:
I understand that this is solved now.

closed 2023-01-24 by kebekus