Akaflieg-Freiburg/enroute #272: Share Qt Creator project

betzuka, 2023-01-22:
I'd like to help you with enroute but I can't get the project setup in creator. Please can you commit your project file, or otherwise share the process you follow to build enroute within creator.

kebekus, 2023-01-22:
THANK YOU FOR YOUR OFFER TO HELP!

I got your message, but I will take a little time to respond because I am underwater with my work. What system do you use for development?

betzuka, 2023-01-22:
No worries you are welcome, it looks like you are mid-port to Qt6 so it may make sense to wait?

I can use either windows or linux, but windows is easier for me as I don't have a non-headless linux box, would end up in a VM. I'm new to Qt Creator hence the problems setting up a project for your codebase. I can code in any language.

kebekus, 2023-01-24:
You should now be able to download the "Enroute" repository from GitHub, right? The next problem is the QtLocation module, which is not present in the latest version of Qt6. 

The app "Enroute Flight Navigation" uses the QtLocation module with the MapLibre plugin for map rendering. The module and the plugin were part of Qt5 but have yet to be officially released for Qt6. The Qt company will likely add QtLocation to the upcoming Qt6.5, but not the module.

The app now uses a self-compiled version of the module and plugin. The compilation is a pain. Here's what to do:

-  Download and install Qt 6.4.2. I recommend the desktop version of Qt. Compiling for Android is possible but still more difficult.
-  Download my patched version of QtLocation from [this repository](https://github.com/Akaflieg-Freiburg/qtlocation). In the repository, switch to the branch "enroute." Look at the shell script "buildscript.sh", which I use to compile QtLocation for Android.
- Inside the Qt 6.4.2/Desktop, you can find the script "qt-configure-module," which is described [here](https://www.qt.io/blog/qt-6-build-system). This wrapper script for CMake sets up all variables required to compile modules.
-  Compile and install. CMake will install the module into the directory where Qt 6.4.2 is installed.
- Now you should be able to compile Enroute.

kebekus, 2023-01-24:
> No worries you are welcome, it looks like you are mid-port to Qt6 so it may make sense to wait?

@betzuka Not really. The port to Qt6 is already done, and I am currently preparing everything for the final release of the app's first Qt6 version.

kebekus, 2023-01-24:
@betzuka If the compilation of QtLocation is too much of a pain, I should point out that several important projects do not require you to compile "Enroute". One such project is a badly needed parser for "Open Air" airspace files, as described in these issues: https://github.com/Akaflieg-Freiburg/enroute/issues/69 and https://github.com/Akaflieg-Freiburg/enroute/issues/230

I guess the parser can be written in a separate project and merged into the app once working. The feature has been requested quite often and should be straightforward to implement, but I had not found the time…

betzuka, 2023-01-24:
> @betzuka If the compilation of QtLocation is too much of a pain, I should point out that several important projects do not require you to compile "Enroute". One such project is a badly needed parser [...]

I will get to building QtLocation, not for a few days though. I want to be able to build enroute as I have a load of my own changes I want to make as well as the roadmap items. I will also have a look at the OA parser.

I'm confused, do you use Qt Creator at all (or just as an editor?) In the wiki it says you do to build android, which is why I asked you share the project file. I haven't used Qt Creator before and just letting it suck up your CMakeLists.txt doesn't work, ends up with some sort of namespace conflict inside Qt which stops Qt compiling. If you don't bother with Creator then I wont bother with it either. 

Do you have a discord for discussion, might work better than here?

kebekus, 2023-01-25:
> I will get to building QtLocation, not for a few days though. I want to be able to build enroute as I have a load of my own changes I want to make as well as the roadmap items. I will also have a look at the OA parser.

Very well.

> I'm confused, do you use Qt Creator at all (or just as an editor?) 

I use the Qt Creator mostly as a convenient, Qt-centric editor. In the past, I found CMake integration in the Qt Creator often rather confusing, and I had difficulties getting reproducible results. After a while, I found it easier to generate binaries with plain CMake and a few shell scripts.

But as I wrote: before you can do anything, you need to compile the QtLocation module with the maplibre plugin.

> Do you have a discord for discussion, might work better than here?

I do not use discord, but we can certainly talk via video. My schedule is super-tight, but I might be available this Friday at 19:30 (Berlin time). Send me a private e-mail.

kebekus, 2023-01-29:
@betzuka I am closing this issue because I feel it does not belong to the "Issues" group. Please get in touch with me anytime by e-mail to discuss things or fix a video chat time.

Your help with programming is GREATLY appreciated.

closed 2023-01-29 by kebekus