Akaflieg-Freiburg/enroute #275: Qt6 6.4.2 - installed from MaintenanceTool - no Qt6Location config

slawekmikula, 2023-01-29:
Was trying to compile latest version of Enroute with Qt6. Installed latest 6.4.2 with all libraries (without source) from the Qt Maintenance Tool (along with the Qt5 version). Providing Qt6 dir gives this error. Cannot find Qt6Location in the installer libs. Any tips for fetching missing dependencies ?

```
-- Could NOT find Qt6Location (missing: Qt6Location_DIR)
CMake Error at CMakeLists.txt:46 (find_package):
  Found package configuration file:

    [...]/qtos/6.4.2/gcc_64/lib/cmake/Qt6/Qt6Config.cmake

  but it set Qt6_FOUND to FALSE so package "Qt6" is considered to be NOT
  FOUND.  Reason given by package:

  Failed to find required Qt component "Location".

  Expected Config file at
  "[...]/qtos/6.4.2/gcc_64/lib/cmake/Qt6Location/Qt6LocationConfig.cmake"
  does NOT exist

  Configuring with -DCMAKE_FIND_DEBUG_MODE=TRUE might reveal details why the
  package was not found.

  Configuring with -DQT_DEBUG_FIND_PACKAGE=ON will print the values of some
  of the path variables that find_package uses to try and find the package.

```

kebekus, 2023-01-30:
@slawekmikula You are right. QtLocation is not part of Qt 6.4.2. You have to compile it for yourself, which is a bit of a nightmare. Things will improve when Qt 6.5 is out and QtLocation is officially included.

If you want to compile the app now, then use the branch "en route" from this repository:

- https://github.com/Akaflieg-Freiburg/qtlocation

I have attached the script to compile the flatpak binary for your convenience.

I hope you are not upset that I am closing this issue now: it's not a bug in Enroute. Please feel free to reach out to me anytime by e-mail.

kebekus, 2023-01-30:
[de.akaflieg_freiburg.enroute.json.txt](https://github.com/Akaflieg-Freiburg/enroute/files/10533452/de.akaflieg_freiburg.enroute.json.txt)

closed 2023-01-30 by kebekus

slawekmikula, 2023-01-30:
yep, not a problem. We have only two kind of issues so it is not request for new feature :) Was expecting this, but anyway reported it here for everyone else to have a solution/info about it. Thanks !