Akaflieg-Freiburg/enroute #276: Unable to import MBTiles

papafox82, 2023-02-03:
Import of MBTiles is not currently working

I have been attempting to import several MBTiles that I have created (via MapTiler and also QGIS) and also other downloaded files that have been known to work in the past for other people.

Steps that I have tried involve copying the file to the phone (either internal memory or SD card), opening the .mbtiles (via a file explorer) into Enroute Flight Navigation. Enroute then opens (if not already open), but nothing further happens. There is no 'hang' time, but the app just continues as before. At no stage does the library list any manually installed maps in the map and data library.

The files that I have been trying with do open successfully in other apps.

I have been primarily using Samsung S9 running Android 10, but have also tried with exactly the same result with a Samsung S20, both running EFR 2.23.00

kebekus, 2023-02-03:
@papafox82 Roger. I plan to investigate the issue over the weekend or next week at the latest. To help the investigation, please upload the relevant MPTiles file [here](https://nextcloud.cplx.vm.uni-freiburg.de/index.php/s/ooyKePqmKzF9JsJ) and send me a short notice once the file is there?

Thank you for your report!

papafox82, 2023-02-03:
Thanks for that. I've uploaded Sydney_VNC_01DEC2022.mbtiles

kebekus, 2023-02-04:
@papafox82 The file arrived, thank you. I can reproduce the issue here and will work on a solution now.

kebekus, 2023-02-07:
Thanks again. I believe I have fixed the issue. The bugfix is contained in version 2.23.90 of the app, which I have uploaded for beta testing to Google Play.

@ papafox82 Please try version 2.23.90 and tell me if things work for you. If you are not a beta tester yet, you can enroll via this link:

- https://play.google.com/apps/testing/de.akaflieg_freiburg.enroute

You can leave the beta channel anytime. If you do not have the time, please tell me so I can continue the publishing process.

Best,

Stefan.

papafox82, 2023-02-07:
All appears to be working great on my end with the update.

Many thanks!

kebekus, 2023-02-08:
> Many thanks!

@papafox82 It's been my pleasure. I do not use that feature, so I would never have noticed without your input. I will give the beta app a few more days and release v2.24.0 on the weekend or early next week.

closed 2023-02-08 by kebekus