Akaflieg-Freiburg/enroute #279: Unable to upload waypoints on Android

antonislarnaca, 2023-02-06:
I exported a way points data file (geojson) from one of my devices and tried to import it to enroute flight navigation of another android device I have using the "share" option. Unfortunately there's no way to do this...
Many thanks

kebekus, 2023-02-06:
@antonislarnaca Thank you for your report! The issue is probably related to [this problem](https://github.com/Akaflieg-Freiburg/enroute/issues/276). I am working on a resolution and hope to have a solution by the end of this week. I will come back to you a.s.a.p.

kebekus, 2023-02-07:
Thanks again. I believe I have fixed the issue. The bugfix is contained in version 2.23.90 of the app, which I have uploaded for beta testing to Google Play.

@antonislarnaca If you have a minute, please try version 2.23.90 and tell me if things work for you. If you are not a beta tester yet, you can enroll via this link:

- https://play.google.com/apps/testing/de.akaflieg_freiburg.enroute

You can leave the beta channel anytime. If you do not have the time, please tell me so I can continue the publishing process.

Best,

Stefan.

antonislarnaca, 2023-02-08:
Hi Stefan,

Awesome. You indeed have fixed the import issue right away. Many thanks for this amazing app. 
I have already become a tester as you recommend. If there is anything I can help with please let me know. I have some knowledge of HTML CSS and some PHP SQL. Unfortunately no programming language...
I fly helicopters for Cyprus police.

All the best and thanks again

Antonis Antoniou
Cyprus 
⁣​

kebekus, 2023-02-08:
I am closing this issue now. Please feel free to re-open if you still experience problems.

Thanks again for your help!

kebekus, 2023-02-08:
> If there is anything I can help with please let me know. I have some knowledge of HTML CSS and some PHP SQL. Unfortunately no programming language...

Thank you for the offer! Let me ponder on this a bit, and I will come back to you.

closed 2023-02-08 by kebekus

kebekus, 2023-02-14:
@antonislarnaca Thanks again for offering to help! How would you like the following projects:

- Translation of the app to your language (Greek, I presume?) and maintaining the translation. At present, the app contains about 1.000 Strings that need a translation. Once this is done, there are usually <10 new strings per month that get added to the app. We use a specialized program to handle translations, so this project does not involve programming.
- (Help with) maintenance of the open aviation databases that this app builds on. The databases "open flightmaps" and "openaip" are run by a few volunteers who need help to maintain and improve the quality of the maps. If you're interested in this, I could get you in touch with the relevant people in this group.