Akaflieg-Freiburg/enroute #282: Export flight routes for use in flight plan

kebekus, 2023-02-26:
The title says it all. Flight plans ask for a route in the following format:

```
WIL DCT LSZW DCT LSTA DCT 4655N00730E DCT LSZW DCT WIL
```

Enroute Flight Navigation could easily export to that format.

fbbln, 2025-04-30:
This seems to be still open, although the mention is closed.
Pushing this up...

fbbln, 2025-05-09:
I'd love to help out and contribute, but I have not worked on Android projects before and I'm rather unfamiliar with C++ (more C and Python).
Yet, if someone is willing / able to give me an intro into tool setup and general structure, I'd be happy to help.

kebekus, 2025-08-03:
Thanks to the help of @charlylima, the feature is included in version 2.34.4, which appeared on Google Play today. As usual, the iOS version will take some further time to appear.

@fbbln Thanks also to you.

Best wishes -- Many Happy Landings -- Stefan.

closed 2025-08-03 by kebekus