Akaflieg-Freiburg/enroute #283: Option "Large Fonts" should increase fonts size in moving map

kebekus, 2023-03-03:
At present, the option "Large Fonts" increases the font size of the GUI elements, but not the font size in the moving map.

sventor, 2023-09-26:
Perhaps you could also consider using a free, open source font made for airliner cockpits like B612: https://b612-font.com/
Only a suggestion, and only if it makes such things easier to solve...!

kebekus, 2023-09-27:
@sventor Thank you for the suggestion! I will consider these fonts. I need to do some expectation management: development is slow because of a lack of manpower. So please do not expect the new fonts to appear very soon…

kebekus, 2024-10-11:
Increasing font size in moving map has been implemented, currently undergoing testing.

kebekus, 2024-10-23:
Implemented in 2.31.14.

@sventor I like your suggestion concerning the B612 font, but I will open a new issue specifically for that.

closed 2024-10-23 by kebekus