Akaflieg-Freiburg/enroute #290: Dialog only scrollable with 4 finger gesture

joachimwe, 2023-03-26:
Hi, 

- Auf meinem Phone lässt sich die Listview der Dialoge nur mit 4 Fingern scrollen.  Im Flug ist das hinderlich - eine einfachere Fingergeste (z.B.1 Finger scroll-Geste) wäre hilfreich.
- Notam Listview - Scrollen bei vielen Einträgen (z.B. LFST)  funktioniert nicht
- z.B. bei längeren Listen z.B. Strasbourg Entzheim (LFST) sind Einträge nicht sichtbar. Wenn Sie wie im Screenshot sauber unten enden, fällt es leider gar nicht auf, dass etwas fehlt. --> vielleicht kann man mit einem Symbol z.B. Pfeil nach unten anzeigen, dass die Liste länger als die darstellbare Fläche ist.

Falls es einen Workaround gibt bzw. es so intendiert ist, danke für eine kurze Rückmeldung.

Tolle Arbeit
Joachim
Samsung Android A71

[image]

kebekus, 2023-03-27:
@joachimwe 

Lieber Joachim,

Danke für die Meldung!

- Zu Punkt 1+2: Ich bin verwirrt, weil ich auf meinem Gerät stets nur mit einer 1-Finger Geste scrolle. Ich wusste gar nicht, dass es eine 4-Finger-Geste gibt. Der Screenshot scheint ein Gnome-Desktop zu zeigen. Auf welcher Platform tritt das Problem denn genau auf?

- Zu Punkt 3: Gute Idee, werde ich implementieren. Ich habe für diesen Punkt einen neuen Eintrag aufgemacht: https://github.com/Akaflieg-Freiburg/enroute/issues/291

Grüße,

Stefan.

joachimwe, 2023-03-27:
Hallo Stefan,

- zu 1) Voraussetzung - lange Listviews z.B. LFST: mein Phone scrollt nur mit 4 Fingerspitzen auf dem Screen (Items sind reine readonly Elemente)
- zu 2) Notams - z.B. bei LFST: Items sind clickable (gelesen bzw. nicht gelesen); hier funktioniert scrollen nur, wenn ich links und rechts am "Rahmen" mit 4 Fingerspitzen (z.B. 2 links, 2 rechts außerhalb der clickable area, rote "Punkte" sind meine Finger :-)) 

Auf einem alten S2 Tablet (Samsung) funktioniert das Scrollen mit einem Finger problemlos.

Ich hoffe, dass dafür eine Lösung gefunden werden kann.

Danke, Joachim

A71 - Android 13 

[image]

[image]

kebekus, 2023-04-05:
@joachimwe Hallo, ich habe inzwischen Version 2.26.7 zu Google Play hochgeladen. Die Version enthält eine Upgrade der zugrundeliegenden Qt-Bibliotheken auf Version 6.4.3. Kannst Du bitte einmal testen, ob Dein Problem damit immer noch auftritt? -- Danke!

kebekus, 2023-04-06:
@joachimwe Der Vorschlag mit dem ListView ist implementiert und wird in der nächsten Version enthalten sein.

kebekus, 2023-05-25:
Closed for inactivity.

@joachimwe Please feel free to re-open if your problems persist.

closed 2023-05-25 by kebekus