Akaflieg-Freiburg/enroute #292: App not working

2dylan1, 2023-03-29:
If i go in to the updated app it doesn't work.. 
[image]

kebekus, 2023-03-30:
Thank you for the report. The message arrived, but I am overwhelmed with my job now. Please give me a few days, and I will get back to you as soon as possible.

2dylan1, 2023-04-05:
> @ 2dylan1 As far as I can tell from the screenshot, the app is working,
> but no map is installed for your region. Please open the main menu, go to
> "Library/Maps and Data," and check what maps are installed. If there *are*
> maps for the region, then I need to investigate the issue more seriously.
>
> Best wishes from London, where we spend a few days with the family.

Hi Stefan

The map, "South Africa" is installed and it is the most recent update of
the app as well as the Flarm Database. It is still not working. I tried to
uninstall the app and re-install it. As well as the map data and it still
does not work. Only thing that is showing is my last known location as in
the screenshot, and the red Notam triangles.

kebekus, 2023-04-10:
@2dylan1 I returned from vacation and will be more responsive now.

The issue that you describe seems serious. At the moment, I am still determining where the problem lies. Can you please do the following?

- What version of the app do you use exactly? Please open the main menu and click "Information/About Enroute Flight Navigation." You can find the precise information at the top of the "ENROUTE" tab there.

- Still on the page "About EFN," please go to the tab "SYSTEM," which gives me information about the system and a list of messages from the operating system. Can you please send me the text? You can extract the text using "Share Info" at the bottom of the screen.

- Next, go back to the moving map (or what should be the moving map) and open "Information/Positioning." This page tells you what the app knows about the device's position. Can you send me a screenshot?

- Finally, return to the moving map and open "Library/Maps and Data." Could you look for the entry for your region (South Africa, I understand)? When installed, a three-dot menu should be at the right of the entry. Open this menu and click on "Info." Can you please send me a screenshot of the info dialog?

Thank you for your patience and your help!

Best wishes,

Stefan.

2dylan1, 2023-04-10:
Hi Stefan

It seems like it's working now. I am still sending you the information as
requested.

kebekus, 2023-04-10:
Thanks. I am closing this issued now, but still looking forward to your data.

closed 2023-04-10 by kebekus