Akaflieg-Freiburg/enroute #325: Add direct-to option for points in route without losing route

mfinnz, 2023-05-04:
It would be great if direct-to could be used on any way point in the current flight plan but without losing the existing route. The logic could be to insert the direct-do within the plan, using the existing direct-to red line. Then after passing the direct-to waypoint the next waypoint in the router sequences as it does already. This is the same behavior that the red line does at the moment when passing a waypoint.

Although the existing red line on the map does show the direct-to and the distance, it is only when within 3nm of the route. Often we need to divert more than 3nm off track, due to weather or traffic, so it would be nice to be able to track direct-to the next waypoint and not lose the saved route.

Hopefully as the direct-to logic and calculations are already there this won't be too difficult to implement.

Thank you

kebekus, 2025-08-03:
@mfinnz The feature that you requested has been implemented in Version 2.34.4, which appeared on Google Play today. As usual, it will take a while for the iOS version to appear.

Thank you for your request! Please keep reporting any issues that you may find -- and please feel free to re-open this issue report if problems persist.

Best wishes -- Many happy landings,

Stefan.

closed 2025-08-03 by kebekus

mfinnz, 2025-08-05:
Great news and it works well thanks. Was amazing to get the update that this is now implemented. Thanks again for your hard work on this.

Happy landings too!

Mark