Akaflieg-Freiburg/enroute #328: Integration with SafeSky

TTC70, 2023-05-23:
Hello,
i believe it would be very useful to have an integration with the app "SafeSky".
It is an app that shows the position of other aircrafts (filtered also by altitude) that use SafeSky and/or have the transponder on.

The url is https://www.safesky.app/en/fonctionnement

This app is already interfaced with other navigation apps: https://www.safesky.app/en/compatibilite

Just an idea.

kebekus, 2023-06-02:
@TTC70 As usual, I apologize for the late reply -- I am on vacation (see the picture attached).

I expect that Enroute already works with the SafeSky app. However, I am unable (or unwilling) to test this because communication with other apps is a "premium feature" requiring a paid subscription. Do you have (or know anyone who has) a premium subscription where we could test that? Alternatively, are you willing to contact SafeSky and ask if they give us a "free premium" for testing and development purposes?

Best,

Stefan.

TTC70, 2023-06-04:
Hey Stefan,
it seems you are having fun with your glider.
In the spirit of sharing, attached my activity of today:

[image]

Regarding SafeSky, I will see what i can do regarding the free beta tester account.
Cheers.

sugacube, 2023-06-17:
i have a safesky premium account and I just tried to couple safesky with enroute selecting the "generic gld90" option (as "enroute" is not among available choices in safesky). This does not (yet) appear to work.

I will post a message on the safesky forum requesting a free premium account for you.

Also, i'd be prepare to 1) fund a premium account for you for a year, en 2) assist in getting this to work, as I really like the enroute app, and would love to have traffic displayed.

kebekus, 2023-06-17:
@sugacube Thank you so much! I am pretty confident that we can make this work…

TTC70, 2023-06-18:
@sugacube thanks!!
@kebekus ...FYI i wrote to SafeSky info describing the situation and asking if they can provide a free premium account for studying purposes. I confirmed that we would like to integrate the information provided by SafeSky with Enroute once Enroute is widely used in Europe.
I will provide more info when they answer.

manolohpm, 2023-06-18:
Hello.

@kekebus I sent a donation after trying your Enroute software which is just what I was looking for. My instructor told me about it.

This donation is enough for a SafeSky license, just in case...

sugacube, 2023-06-20:
@kekebus,
I got feedback from the skysafe forum that you can contact the developer directly via mail to get help.
I don't want to drop his email here, any way i can send it to you ? (you can also look on the skysafe forum yourself)

@manolohpm  i didn't know you could do a donation, that information is hidden very well 😀

sugacube, 2023-06-23:
@manolohpm yes, i eventually found it. But honestly, who still reads a manual nowadays ?

kebekus, 2023-06-24:
@sugacube @manolohpm @TTC70 Thanks for all your feedback! I am a little slow to reply these days, but the e-mail of @manolohpm did reach me.

I understand from all your feedback that SafeSky integration is important to you, so I will prioritize it accordingly. I am confident I can make things work quickly when I get a full version of SafeSky. I will contact the SafeSky developers as @manolohpm suggested in his mail.

I will keep you updated!

Best,

Stefan.

sugacube, 2023-06-27:
@manolohpm 
@anyone reading this thread

Based on feedback from Safesky, the Enroute-Safesky integration seems to work already if you select following in safesky "Traffic Sharing" :
- Enable Traffic Sharing
- Generic GDL90
- Force to use SafeSky as a GPS source

manolohpm, 2023-06-27:
Hi.

@kekebus Thank you for your answer

@sugacube Thank you. Making some tests here.

1) Testing on a 9th gen iPad without GPS/SIM card. SafeSky and Enroute running in the same iPad. Traffic is shown, but position is read from the ISP provider. This configuration won't work while flying.

2) Android phone (GPS) running SafeSky and previous iPad running Enroute, and connected to android phone access point. Traffic is not shown. Red plane is shown on the enroute screen.

I don't know if sharing GDL90 data through access point is possible.

sugacube, 2023-06-28:
> I don't know if sharing GDL90 data through access point is possible.

Don't know either, I haven't tried that, but I suppose it should. Maybe ask on safesky forum, maybe try by explicitly configuring ip address.

If you try safesky/enroute running simultaneous on your phone, (safesky in background) , that seems to work.

My tablet has gps, so i don't face the same issue. (fyi aldi sells 200 € tablets with built-in gps/3g 😄)

kebekus, 2023-06-29:
@TTC70 @sugacube @manolohpm Thanks to all of you for your help. Now that SafeSky is confirmed to work, I have added a [section to the manual](https://akaflieg-freiburg.github.io/enrouteText/manual/02-steps/safesky.html). The next version of **Enroute** will have appropriate links on the 'Settings' page and on the 'Traffic Info' page.

I am closing this issue now, feel free to re-open is problems show up.

closed 2023-06-29 by kebekus