Akaflieg-Freiburg/enroute #329: Building the project fails on Arch Linux

mgw93, 2023-05-23:
**Describe the bug**
Building the project fails on Arch Linux

**To Reproduce**
Attempt to build the linux version according to [the instructions](https://github.com/Akaflieg-Freiburg/enroute/wiki/Compilation-and-additional-cmake-targets).
A number of issues will come up.
I attempted to solve each of them, but got stuck at a linker error.

**Expected behavior**
Build succeeds

**Desktop (please complete the following information):**
 - OS: Arch Linux

**Additional context**
A number of issues came up:

- CMake failed with an error message about 
`CMake Error at src/CMakeLists.txt:497 (find_path):
  Could not find Qt6_TRANSLATION_DIR using the following files: qtbase_de.qm`
I was able to solve this by adding `HINTS /usr/share/qt6/translations` to the CMakeLists.txt file. There may however be a more generic way. I have little experience with CMake.
- CMake failed with an error message about `CMake Error: The INTERFACE_QT_MAJOR_VERSION property of "Qt5::Core" does
not agree with the value of QT_MAJOR_VERSION already determined
for "enroute".` I was able to fix this by deleting `find_package(QT NAMES Qt6 Qt5 COMPONENTS Network REQUIRED)` from 3rdParty/KDSingleApplication/CMakeLists.txt, and by calling cmake as `cmake  -DQT_VERSION_MAJOR=6 ..` I am however not sure whether this introduced any new issues.
- Compiling after successfully running CMake resulted in an error because the TileServer class couldn’t inherit from QAbstractHttpServer. Overriding the handleRequest and missingHandler functions failed due to a mismatch between the function signatures. Editing the code in src/geomaps/TileServer.cpp to 
```
bool GeoMaps::TileServer::handleRequest(const QHttpServerRequest& request, QHttpServerResponder &responder)
{
    auto path = request.url().path();
    auto pathElements = path.split('/', Qt::SkipEmptyParts);
    //auto responder = makeResponder(request, socket);

…
```
and
```
void GeoMaps::TileServer::missingHandler(const QHttpServerRequest& request, QHttpServerResponder &&responder)
{
    //auto responder = makeResponder(request, socket);
    responder.write(QHttpServerResponder::StatusCode::NotFound);
}
```
solved the compilation issue.
- Now the linker fails with the message 
```/usr/bin/ld: qrc_ressources.o: copy relocation against non-copyable protected symbol 
qt_resourceFeatureZstd@@Qt_6' in /usr/lib/libQt6Core.so.6.5.0
collect2: Fehler: ld gab 1 als Ende-Status zurück
make[2]: *** [src/CMakeFiles/enroute.dir/build.make:3945: src/enroute] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:786: src/CMakeFiles/enroute.dir/all] Fehler 2
make: *** [Makefile:136: all] Fehler 2
```
I personally have no idea how to debug this further. It may also be an issue that I introduced with my earlier changes.

mgw93, 2023-05-23:
I now noticed that the linker error goes away when building with clang instead of gcc.
The built app now starts. It however does not show a map, even after downloading map data.
[image]
Possibly this is related to the change I needed to make to the TileServer class.

kebekus, 2023-05-25:
Dear @mgw93,

Thank you for your report. Compiling the app is difficult because it requires the QtLocation module and the high-performance MapLibre plugin. Currently, these exist only for Qt6.4.3, so you must compile the app with Qt6.4.3. The app you compiled with Qt6.5 does not show a map because the MapLibre plugin is missing.

- You can find binaries for QtLocation+MapLibre for Qt6.4.3 here: https://github.com/maplibre/maplibre-native-qt/releases/tag/v2.1.0

- Alternatively, you can compile QtLocation+MapLibre for Qt6.4.3 yourself using the code from this repository: https://github.com/Akaflieg-Freiburg/qtlocationstatic

You might find it instructive to look at the following scripts, which can compile Enroute successfully:

- the GitHub actions in the "Enroute" repository

- the flatpak manifest in https://github.com/Akaflieg-Freiburg/qtlocationstatic

I am looking forward to QtLocation+MapLibre for Qt6.5.x. Once this is out, compilation will become much easier.

Best,

Stefan.

kebekus, 2023-05-25:
@mgw93 Thanks again for your report. I am closing this issue now. Feel free to re-open if your problems persist.

Would you be willing to help with the application, perhaps by writing a "How-to-Compile" in the Wiki?

closed 2023-05-25 by kebekus