Akaflieg-Freiburg/enroute #331: Airport location

P4U1x, 2023-06-11:
Hey, I found an airport in Denmark (Kerteminde), based on enroute the airport is near the Kerteminde Fjord, but in reality (google Maps), the airport is further down south near the street 315. Is it an error of the app or off the map data that you use? 
 Coordinates according to enroute: 55,44713/10,59868

Coordinates according to google maps:
55,4184729/10,6706948

(And by the way I love enroute, thanks for all the work you do <3)

Have a nice week!!

kebekus, 2023-06-16:
@P4U1x Thank you for your report. We obtain airfield information from openaip.net, and the data needs to be corrected there. Once the correction is accepted at openaip, it will automatically appear in **Enroute Flight Navigation** within 24h.

Peter Kemme, a regular contributor to openaip, has volunteered to help our users. Can you please get in touch with Peter? I should not post his e-mail address here, but you can find it in the app (go to "Main Menu/Bug Report"). Alternatively, send me an e-mail.

I am closing this issue now. Please feel free to re-open or add comments, I will read them.

Best,

Stefan.

closed 2023-06-16 by kebekus

P4U1x, 2023-06-16:
I'll write him. Thanks!!