Akaflieg-Freiburg/enroute #332: Cannot open mbtiles file with iOS version

christianhbeck, 2023-06-20:
**Describe the bug**
In iOS enroute is not associated with the mbtiles file type. This i cannot open mbtiles filed

**To Reproduce**
Steps to reproduce the behavior:
1. Go to files
2. Click on mbtiles file

**Expected behavior**
On android clicking the file gives the option to send to enroute

**Smartphone (please complete the following information):**
 - Device: [xs]
 - OS: [latest]

kebekus, 2023-06-20:
@christianhbeck Thank you for the report. I can confirm the issue. We're working on the problem and will reply as soon as possible.

schneisi, 2023-06-20:
Fixed with the latest Version 2.27.0 (7). The update is available via TestFlight.
@christianhbeck Please confirm it works on your device.

christianhbeck, 2023-06-20:
Hello,
I have installed the new version and it works perfectly. Thank you very much for the fix, the iOS version and this great project in general!

regards,
Christian

kebekus, 2023-06-21:
@christianhbeck Thank you for the feedback.
@schneisi Thank you for the work.

closed 2023-06-21 by kebekus