Akaflieg-Freiburg/enroute #333: PingUSB ADS-B receiver doesn't show traffic altitude

mfinnz, 2023-06-30:
Hi,

I am using the uAvonix PingUSB ADS-B receiver, but although it shows traffic information, no altitude or speed info is displayed. It only shows callsign, position, and type. I have tried this on 2 x Android devices and on the beta IOS version on an iPad with the same result.

The device is using GDL90, connected via Wifi, and it is made by the same company as the SkyEcho. I have also tried OzRunways and the traffic information is displayed, including the height and speed of traffic through ADS-B. It works on all 3 devices.

Screenshot of Enroute:

[image]

Screenshot of OzRunways:

[image]

Is this expected behavior for a GDL90 protocol device? I understand it can be difficult to troubleshoot without the device, but if this is expected behavior then I won't worry about testing further. Let me know if you need any more information or testing.

Thank you

Mark

kebekus, 2023-06-30:
Dear Mark,

Thank you for your report! The behavior that you experience is the expected one. The app that claims to show the relative vertical distance is giving you dangerously misleading information.

Detailed explanation:

First, I have deliberately chosen not to display the precise speed because the relevant information is already visible in the 5-minute-projected trajectory. Adding a numerical value for speed wastes screen space and makes the display harder to understand. Every data item I include adds a second to the time the pilots need to understand the information.

Much more importantly: relative height. As you can see in the GDL90 specification, the traffic height reported via GDL90 is a PRESSURE height, referenced to 29.92 inches Hg. Your own pressure height is, however, unknown because the pingUSB has no pressure sensor (and even if it had a sensor, that would have to be connected to the static pressure port of your aircraft). Your cell phone or tablet can use GPS to find its GEOMETRIC height, but depending on the weather, the PRESSURE height and GEOMETRIC height values can differ dramatically. Bottom line: the two MUST NOT be compared to estimate the relative height between your aircraft and traffic. THIS IS MISLEADING AND DANGEROUS!

**Enroute Flight Navigation** will show the relative height of traffic as soon as you connect to a device that reports the ownship pressure altitude. The low-end pingUSB device does give this crucial piece of information. IMHO, pingUSB should not be used in aviation.

I understand this is not what you wanted to hear, and I am sorry I have no better news. I am closing this issue now, but I will monitor this discussion thread. Please feel free to come back to me with questions or concerns. Please re-open the issue when appropriate.

Best wishes,

Stefan.

closed 2023-06-30 by kebekus

kebekus, 2023-06-30:
Update: I added a warning and an explanation to the manual:

- https://akaflieg-freiburg.github.io/enrouteText/manual/04-appendix/technical.html#pingusb

mfinnz, 2023-06-30:
Hi Stefan,

Thank you for your fast and detailed reply, that all makes sense. I had read your note about altitude reporting in the manual and wondered if that was the case why it wasn't shown in the traffic info. That confirms it and I see you updated manual already, so that's fantastic.

I agree it is indeed a trap, and although small pressure differences are not so much of a hazard, when you are talking extreme QNH that can be very dangerous.

Thanks for all your hard work on the app.

Regards from New Zealand,

Mark

kebekus, 2023-06-30:
Dear Mark, thanks for your understanding! I often fly in the Swiss Alps. On cold winter days and high elevations, the difference between geometric and barometric altitudes can easily be 300m. Given that we have only 500ft vertical separation (!) between GA airspace and the Swiss military fighter jets, this **is** an issue…