Akaflieg-Freiburg/enroute #334: Text in waypoint description dialog is cut off

kebekus, 2023-07-02:
Simon has sent me the following screenshot (from iOS, 2.27.1)

[image]

kebekus, 2023-07-03:
I cannot reproduce the problem. Here's what it looks like for me.

[image]

kebekus, 2023-07-03:
Simon:  tritt auf einem iPhone 8 auf, auf meinem 14 nicht. 

App neu starten -> Flugplatz öffnen => Kein Bug
Schließen -> Dialog erneut öffnen ⇒ Bug
Landscaper drehen ⇒ nein Bug
Zurück Portrait drehen ⇒ kein Bug
Dialog erneut öffnen ⇒ Bug

kebekus, 2024-02-13:
@schneisi Existiert das Problem noch?

schneisi, 2024-02-13:
Kann ich mit einem aktuellen 2.30.X Build auf dem betroffenen Gerät nicht mehr reproduzieren

kebekus, 2024-02-14:
@schneisi Besten Dank, ich schließe den Fall dann.

closed 2024-02-14 by kebekus